esphome/home-assistant-voice-pe #322: [REQUEST] Add ”audio synchronizer” feature to Audio Media Player for synchronization between multiple ESPHome media players for synchronized multi-room music playback

Hedda, 2025-02-10:
Feature request look into adding some kind of ”audio synchronizer” feature to support Audio Media Player synchronization between multiple ESPHome media players for synchronized multi-room music playback.

The goal is to achieve synchronous multiroom audio playback for music using client-server audio player, where all clients (media players) are time synchronized with the server to play perfectly synced audio.

Multiroom audio (a.k.a. distributed audio) with a such ”audio synchronizer” feature for synchronizing music playback on multiple speakers is a key feature of competing voice assistant / smart speaker echosystems.

Suggest maybe look into [rwrozelle’s Audio Media Player code for ESPHome](https://github.com/rwrozelle/audio-media-player) which has an implementation that allow other audio media player members to join a group and then the group members will turn off, turn on, set volume, and play the same media as leader. Synchronization is attempted by telling leader and group members to start media at the same time. Uses the Time component in the sntp platform to assume that chips have the same time. See:

- https://github.com/rwrozelle/audio-media-player

PS: Spotted that ([kahrendt](https://github.com/kahrendt)) already implemented a lot of related new media player features which was just now merged into upstream ESPHome, like:

- https://github.com/esphome/esphome/pull/8164

  * *Adds a playback duration callback to the parent speaker component (and implements it in the i2s_audio speaker component). This precisely tracks the audio duration written to the DMA buffers to potentially enable synced playback in a future update.*

  * *Adds pausing to the parent speaker component (and implements it in the i2s_audio speaker component). This will reduce latency when pausing the media player when several virtual speakers are stacked.*

and

- https://github.com/esphome/esphome/pull/8171

  * *PR9 adds a new speaker media player component that sends audio to another ESPHome speaker component. It supports two pipelines, one for announcements and one for regular media. Each pipeline requires a unique output speaker. Use a mixer speaker from https://github.com/esphome/esphome/pull/8170 to create two source speakers that are mixed and sent to another speaker component.*

  * *To increase the number of audio formats available and to minimize the load on the ESP32, you can use HA to transcode audio automatically before it is sent to the device. By default, it transcodes to FLAC, which uses the CPU the least. You can set it to transcode to MP3, WAV, or disabled. You can specify the sample rate and number of channels for the transcoding, but by default it will match the output speaker's setting. If you disable HA transcoding, then the media player only supports FLAC, MP3, and WAV files.*

WhiteSockedDancer, 2025-03-02:
The use of Music Assistant, where you can group devices and use them in MA or HA should also be considered. :)

kahrendt, 2025-05-22:
As mentioned in your other issue, this is a more general ESPHome feature request, and we are in the early stages of adding synced audio support.

closed 2025-05-22 by kahrendt