esphome/home-assistant-voice-pe #324: Not compiling on esphome beta

tbrasser, 2025-02-12:
```yaml
substitutions:
  name: home-assistant-voice-0905d2
  friendly_name: Eetkamer Voice
packages:
  Nabu Casa.Home Assistant Voice PE: github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml@dev
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
```
INFO ESPHome 2025.2.0b1 INFO Reading configuration /config/esphome/home-assistant-voice-0905d2.yaml... INFO Updating https://github.com/esphome/home-assistant-voice-pe@dev WARNING GPIO0 is a strapping PIN and should only be used for I/O with care. Attaching external pullup/down resistors to strapping pins can cause unexpected failures. See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins WARNING GPIO3 is a strapping PIN and should only be used for I/O with care. Attaching external pullup/down resistors to strapping pins can cause unexpected failures. See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins Failed config light.esp32_rmt_led_strip: [source /data/packages/f1018e04/home-assistant-voice.yaml:556] platform: esp32_rmt_led_strip id: leds_internal pin: GPIO21 This feature is not available for the IDF framework version 5. rmt_channel: 1 num_leds: 12 rgb_order: GRB chipset: WS2812 default_transition_length: 0ms power_supply: led_power
```

aryajuanda, 2025-02-13:
the `rmt_channel: 1` is not available in new idf framework, need to be removed

tbrasser, 2025-02-13:
after fixing that (using !remove) I get:

```
Traceback (most recent call last): 
File "/usr/local/bin/esphome", line 8, in <module> sys.exit(main()) ^^^^^^ 
File "/esphome/esphome/__main__.py", line 1049, in main return run_esphome(sys.argv) ^^^^^^^^^^^^^^^^^^^^^ 
File "/esphome/esphome/__main__.py", line 1036, in run_esphome rc = POST_CONFIG_ACTIONS[args.command](args, config) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ 
File "/esphome/esphome/__main__.py", line 503, in command_run exit_code = write_cpp(config) ^^^^^^^^^^^^^^^^^ 
File "/esphome/esphome/__main__.py", line 212, in write_cpp generate_cpp_contents(config) 
File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents CORE.flush_tasks() 
File "/esphome/esphome/core/__init__.py", line 674, in flush_tasks self.event_loop.flush_tasks() 
File "/esphome/esphome/coroutine.py", line 246, in flush_tasks next(task.iterator) 
File "/esphome/esphome/__main__.py", line 204, in wrapped await coro(conf) 
File "/esphome/esphome/components/audio/__init__.py", line 121, in to_code cg.add_library("esphome/esp-audio-libs", "1.1.1") 
File "/esphome/esphome/cpp_generator.py", line 603, in add_library CORE.add_library(Library(name, version, repository)) 
File "/esphome/esphome/core/__init__.py", line 748, in add_library raise ValueError( ValueError: 
Version pinning failed! 
Libraries esphome/esp-audio-libs@1.1.1 and esphome/esp-audio-libs@1.0.0 requested with conflicting versions!
```

goofy1988, 2025-02-13:
Hey there,

i have the same issue.

Ive got the same error message when compiling. My try to remote the LED and Add a new one without the rmt_channel: 1 gave me the same error as you mentioned:

INFO ESPHome 2025.2.0b2
INFO Reading configuration /config/esphome/home-assistant-voice-090fa9.yaml...
INFO Updating https://github.com/esphome/home-assistant-voice-pe@dev
WARNING 'wz_nabu_voiceassistant': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1049, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 674, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/audio/__init__.py", line 121, in to_code
    cg.add_library("esphome/esp-audio-libs", "1.1.1")
  File "/esphome/esphome/cpp_generator.py", line 603, in add_library
    CORE.add_library(Library(name, version, repository))
  File "/esphome/esphome/core/__init__.py", line 748, in add_library
    raise ValueError(
ValueError: Version pinning failed! Libraries esphome/esp-audio-libs@1.1.1 and esphome/esp-audio-libs@1.0.0 requested with conflicting versions!

_____________________________________________

This is the remove and readd part in my code:

light:

  - id: !remove leds_internal
  
  # Hardware LED ring. Not used because remapping needed
  - platform: esp32_rmt_led_strip
    id: leds_internal
    pin: GPIO21
    #rmt_channel: 1
    num_leds: 12
    rgb_order: GRB
    chipset: WS2812
    default_transition_length: 0ms
    power_supply: led_power

rccoleman, 2025-02-17:
Simply bumping the `esp-audio-libs` dependency to 1.1.1 in https://github.com/esphome/home-assistant-voice-pe/blob/dev/esphome/components/nabu/media_player.py isn't enough (generates more errors during the compilation, so hopefully there's a PR out there to make the Voice PE config functional with the new ESPHome beta.

[esphome-2025.2-beta-voice-pe.txt](https://github.com/user-attachments/files/18832481/esphome-2025.2-beta-voice-pe.txt)

matze19999, 2025-02-19:
This now also happens on ESPHome Stable. Can we only use esphome/esp-audio-libs@1.0.0?

Magicpulse, 2025-02-19:
Upgraded ESPHome to 2025.2.0 and I confirm this happens.

vhamburger, 2025-02-19:
Same error here - seems the yaml / firmware needs some fixing please! (pretty please?)

kahrendt, 2025-02-19:
The yaml config has been updated in the last hour or two with the required change to make it build on ESPHome 2025.2.0. A new beta will be published soon as well if you haven't taken control of the firmware. 

Thanks for reporting!

closed 2025-02-19 by kahrendt

rccoleman, 2025-02-19:
Working fine for me, thanks!