esphome/home-assistant-voice-pe #325: Request keep listening

maxtrin87, 2025-02-13:
Would be great if it can keep listening sometimes ai asks a question and I want to answer back

kosmaluk, 2025-02-16:
The only problem is downloading the content of this response to a file to analyze it. 

I lost a lot of time looking for a way to save the agent's response, but I found a good solution. 

Finally, for the time being I took control and added sensors in configuration.yaml in ESP Builder, but this is a temporary solution. 

Maybe someone knows how to save the agent's response?

aryajuanda, 2025-02-17:
after ask in the community and reading the documentation its easy to create logit to make assistant listening again after answering the question, we can get plain text of the response from the yaml code 
well documented here : https://esphome.io/components/voice_assistant.html

we can get the response from `on_tts_start` and we can add logic with it.. i have made some test with a generic esp32s3 and its works, it will listen again if we want to (example : detecting if need answer and call `voice_assistant.start` in the `on_end` )

itispip, 2025-02-20:
> after ask in the community and reading the documentation its easy to create logit to make assistant listening again after answering the question, we can get plain text of the response from the yaml [...]

Is this only possible after adopted to ESPHome? But Home Assistant official documentation not ecouraging the adoption

kosmaluk, 2025-02-21:
the official documentation was written a few months ago. Today you can easily take over the code and make changes without losing the ability to update. A new Home Assistant Voice Firmware update sensor has been added. You can add your own media_player or create start/stop wakeword buttons and in automations make a continuation of the conversation, dialogflow, etc.

aryajuanda, 2025-02-28:
> > the official documentation was written a few months ago. Today you can easily take over the code and make changes without losing the ability to update. A new Home Assistant Voice Firmware update sensor has been added. You can add your own media_player or create start/stop wakeword buttons and in automations make a continuation of the conversation, dialogflow, etc.
> 
> Thanks. can you share how you get the response from on_tts_start, so that a logic can be apply to determine "restart listen" or "stop listen"?
> 
> And why this logic should be on_tts_start rather than on_tts_end, or even on_stream_end? 

Because the reply text in available as x param on_tts_start

So i create logic to rean the respons and flag to a new global parameter if need reply

on_end i check the global param and start the voice_assistant.start again if need reply

itispip, 2025-02-28:
> > > the official documentation was written a few months ago. Today you can easily take over the code and make changes without losing the ability to update. A new Home Assistant Voice Firmware update sensor has been added. You can add your own media_player or create start/stop wakeword buttons and in automations make a continuation of the conversation, dialogflow, etc.
> > 
> > 
> > Thanks. can you share how you get the response from on_tts_start, so that a logic can be apply to determine "restart listen" or "stop listen"?
> > And why this logic should be on_tts_start rather than on_tts_end, or even on_stream_end?
> 
> Because the reply text in available as x param on_tts_start
> 
> So i create logic to rean the respons and flag to a new global parameter if need reply
> 
> on_end i check the global param and start the voice_assistant.start again if need reply
> 
> here is my [DIY VA](https://github.com/aryajuanda/ha-voice/blob/main/smart-speaker.yaml)

Thanks. That makes a lot of sense.

ajfisher, 2025-03-08:
Am not sure I totally follow the work around to this, regardless though it feels like having a simple time out after a response by the VA where it drops back into listening mode would allow you to keep a conversation flowing. I'm really impressed by the mic ability of the PE but this gets old really quickly:

"Okay nabu"

"Turn on the lights"

"I've turned on the lights"

"Okay nabu"

"Close the blinds"

"I've shut the blinds"

Etc. 

I know you can do a lot with scenes (and I have which is good) but I'm now in this weird scenario when you say 'turn on the lounge lights and turn off the kitchen lights and turn on the tv and shut the lounge blinds and give me the weather forecast for tonight" just leveraging the fact that the Open AI integration will parse all of that into appropriate commands. But it's definitely much more complex from a user standpoint than keeping the conversation open for 10 seconds so you can issue another instruction....

kahrendt, 2025-04-02:
This is now possible with Home Assistant 2024.4 and an LLM. If the LLM text response ends in a question mark, then it will automatically start the pipeline again without requiring the wake word. VPE firmware 25.3.0 added the corresponding support for this.

closed 2025-04-02 by kahrendt

ajfisher, 2025-04-03:
This is great that this is in the latest update - super keen to try this out over the next couple of days.