esphome/home-assistant-voice-pe #327: 24.12.3 will connect to Wifi, nothing afterwards will

RobMeades, 2025-02-19:
Hi there.  I got my Voice Assistant PE box today and immediately connected it to my Wifi, all initially good.

However, once connected, Home Assistant updated the FW on the device (to 25.1.0 I think) and, after that the box refused to connect to Wifi.  I've tried loading a few different FW releases (25.1.1, 24.12.8 and then 24.12.3) but 24.12.3 is the only one that works, however of course the FW is immediately updated by Home Assistant and all is lost.

The symptom of the failure is that the new FW appears to have been loaded by Home Assistant, the dialog box says "resetting device", the device ring lights up bright white for quite a long time (maybe a minute), then it goes to sort of a random pattern and the device is unavailable in Home Assistant.  Attached find a log of what the device does from boot when in this state, along with a log of what it does with the 24.12.3 when Wifi connectivity works (if I skip the "connect to Home Assistant" step).

My Wifi network is 2.4 GHz, WPA2, nothing unusual (I disabled my separate 5 GHz Wifi network in case of HW issues).

Is there a way to nail the FW version to 24.13.2 in Home Assistant so that I can at least get working?

[esp-web-tools-log-connect-fails.txt](https://github.com/user-attachments/files/18869595/esp-web-tools-log-connect-fails.txt)
[esp-web-tools-log-connect_success.txt](https://github.com/user-attachments/files/18869669/esp-web-tools-log-connect_success.txt)

RobMeades, 2025-02-19:
Possible duplicate of #297 but since that was raised a month ago and doesn't seem to have been looked at I'm going to leave this one here (with the logs attached).

RobMeades, 2025-02-20:
Checking further on this, my router is expecting a connected device to use  "WPA Personal", "TKIP/AES" security.

In the "success" log above, you will see that ESPHome says:

```
connected with Meades18, aid = 19, channel 1, BW20, bssid = 2c:56:dc:4f:65:20
security: WPA-PSK, phy: bgn, rssi: -47
```

...whereas in the "fail" log it says:

```
WiFi Connecting to 'Meades18'...
Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
Event: Disconnected ssid='Meades18' reason='Probe Request Unsuccessful'
```

...so are though the later ESPHome versions guessing the wrong Wifi security type...?

RobMeades, 2025-02-20:
I've added debug to the code of [esphome](https://github.com/esphome/esphome) and tracked down the issue.  There seem to be two things wrong.

The main one is that here:

https://github.com/esphome/esphome/blob/bf739506c352fd24881f32aa79fef6a2e81298a8/esphome/components/wifi/wifi_component_esp_idf.cpp#L304-L308

...if a password has been set, the authentication threshold for fast scans, which will be used in this case, is being set to `WIFI_AUTH_WPA_WPA2_PSK`.  This means that the network must indicate support for BOTH `WPA` and `WPA2`, i.e. a network supporting just `WPA` will be ignored, see the order of values in the `enum`:

https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/components/esp_wifi/include/esp_wifi_types_generic.h#L75-L100

My router does support `WPA2`, I just set it to `WPA` because I have devices that will never support `WPA2`.

Changing this code to set `WIFI_AUTH_WPA_PSK` instead, which _should_ fix the problem, doesn't though because, a few lines further down the authentication threshold which was so carefully set is being over-written with `WIFI_AUTH_OPEN` in any case:

https://github.com/esphome/esphome/blob/bf739506c352fd24881f32aa79fef6a2e81298a8/esphome/components/wifi/wifi_component_esp_idf.cpp#L346

This means that the threshold  [esphome](https://github.com/esphome/esphome) sets is, actually, "open", it will always connect to anything in fast scan mode, whatever password is set.  However, if you look at the comment here:

https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/components/esp_wifi/include/esp_wifi_types_generic.h#L359

> Note: In case this value is not set and password is set as per WPA2 standards (password len >= 8), it will be defaulted to WPA2 and device won't connect to deprecated WEP/WPA networks. Please set auth mode threshold as WIFI_AUTH_WEP/WIFI_AUTH_WPA_PSK to connect to WEP/WPA networks

My WiFi password is, of course eight or more characters in length, so setting a threshold of `WIFI_AUTH_OPEN` actually causes WiFi to specifically NOT connect to my `WPA`-authenticated WiFi network, despite the good intentions of [esphome](https://github.com/esphome/esphome) .  That comment [was added by Espressif in late October last year](https://github.com/espressif/esp-idf/commit/a68cf1418b25396649344fdf31eae860d992b0f9), so it seems plausible that this change in ESP-IDF behaviour came about around then, not far off when the problem appeared in [esphome](https://github.com/esphome/esphome) (see also #297).

It is possible that I can make things work by modifying my WiFi network to indicate support for both `WPA` and `WPA2`: it has been the way it is for 20 years for some reason or other, so that _may_ not be possible, I may have old and irreplaceable devices that will not play ball if support for `WPA2` is signalled.

In any case, it seems to me that [esphome](https://github.com/esphome/esphome) is not doing what it means to do; I will propose a PR.

garbled1, 2025-04-12:
I have the same issue as above.  WPA network would not allow device to connect.  Going to attempt "mixed" and see if it breaks my other devices or not...