esphome/home-assistant-voice-pe #330: Update deprecated assist_in_progress binary sensor

brystmar, 2025-02-20:
Since initializing my Voice PE device a few weeks ago, HA has prompted me to fix this repair issue:

> **Repair issue**
> Warning ⸱ Reported by ESPHome
> 
> This stops working in version 2025.4. Please address before upgrading.
> 
> The ESPHome in progress binary sensor binary_sensor.kitchen_voice_assistant_assist_in_progress is deprecated.
> 
> Migrate your configuration to use the corresponding assist_satellite entity and then click SUBMIT to disable the in progress binary sensor and fix this issue.

Can this be addressed through an OTA update?

I'm running HA 2025.2.4, HAOS v14.2 on a dedicated NUC.  ESPHome Device Builder Addon v2025.2.0

jf-uu, 2025-02-20:
Duplicate of #287. The entity is disabled by default and is deprecated (planned to be removed) - re-disable it to fix the issue.

kahrendt, 2025-05-22:
As mentioned, just leave it disabled to fix it, and eventually the entity will be removed after the deprecation period.

closed 2025-05-22 by kahrendt