esphome/home-assistant-voice-pe #334: Allow use of `openWakeWord`

RobMeades, 2025-02-20:
It was a surprise to me that Voice Assistant does not allow use of `openWakeWord`: I had purchase it for this very purpose, to replace an M5 Atom.

I don't care if this is slower, that's not a problem, but I am never likely to use someone-else's wake-word with my local house AI, that is, after all, the half the point of it being _local_.

Anyway, it seems trivial to do, in that I was able to take the [modified yaml](https://gist.github.com/jlpouffier/41351187e2f6f94e797382a658702433) that @jlpouffier kindly posted several months ago and apply it to the most recent version of this repo (on a fork which can be found [here](https://github.com/RobMeades/home-assistant-voice-pe)).  These modifications appear to make my device work fine with `openWakeWord`; not tested for hours yet, but looking good so far.

Could this repo be updated to support it please, then I won't need to hack the device to get what I need.

Or I could make a PR with the contents of the fork if that's easier?

aryajuanda, 2025-02-20:
why dont you train your own micro wake word?

RobMeades, 2025-02-20:
I did try but the word didn't work, whereas the `openWakeWord` one that I trained did.

aryajuanda, 2025-02-21:
or you can try add option to select wake word location like in https://github.com/esphome/wake-word-voice-assistants

with `select:
  - platform: template
    entity_category: config
    name: Wake word engine location
    id: wake_word_engine_location
    optimistic: true
    restore_value: true
    options:
      - In Home Assistant
      - On device
    initial_option: On device
    on_value:
      - if:
          condition:
            lambda: return x == "In Home Assistant";
          then:
            - micro_wake_word.stop
            - delay: 500ms
            - lambda: id(va).set_use_wake_word(true);
            - voice_assistant.start_continuous:
      - if:
          condition:
            lambda: return x == "On device";
          then:
            - lambda: id(va).set_use_wake_word(false);
            - voice_assistant.stop
            - delay: 500ms
            - micro_wake_word.start`

and and some logic in different location to read the parameter

RobMeades, 2025-02-21:
Interesting: I don't think I'm ever likely to want the on-device wake-up words but it is fun to see all of the YAML logic.  Not sure about this "no programming skills" thing, just programming in an alien language :-).

Rjohnson3452, 2025-02-22:
Hey Rob, just wanted to say thank you! I've been searching for a way to get custom wake words since the PE showed up. I made a custom wake word and used it with atom echos and was very disappointed when the PE shipped without them. I'm sure the developers are working hard getting things ironed out on PE's but leaving custom wake words out reminded me of google or amazon. I've used your code on 2 PE's so far and plan on buying another 2. I appreciate the hard work!

RobMeades, 2025-02-22:
@Rjohnson3452 NP, it was @jlpouffier who did the work really, I just pasted things together :-).

Felitendo, 2025-02-24:
@Rjohnson3452 even Amazon (Alexa) has custom wake words

jf-uu, 2025-02-24:
@Felitendo not really sure that's a fair comparison.

Felitendo, 2025-02-24:
@jf-uu I was referring to @Rjohnson3452 where he wrote:
> but leaving custom wake words out reminded me of google or amazon.

ThatProgrammerr, 2025-03-24:
I know I may seem silly here, but how exactly does one even start with implementing a custom micro wake word? I'm sick of 'ok nabu' and it honestly makes me not want to use my Voice PE. Looking at the code sample above, it's not clear where this would go, or if I'd need to maybe build my own firmware?

inertia666, 2025-04-18:
The limited wake words is a huge oversight and must be a good reason for it. Using a female voice assistant and having to call it Mycroft feels really weird. Having to say "ok" or "hey" as part of the wake word is super annoying.

stroodle96, 2025-04-18:
@RobMeades  great solution.  Your fork has been working well for me. Are you able to update it to include support for the new start conversation feature that was added in 2025.4? The upstream repository has the support, so i think it just needs to be synced to yours.

darrenjackson72, 2025-04-21:
@aryajuanda Could you please give me a few short instructions on where to implement your code please? Thanks!

mcbutterpants007, 2025-04-21:
This seems fairly strait forward install, but I agree that maybe a tutorial on how to do this would be great.  I don't want to nuke my PE. .........OR the makers of this device can can just make this an option.  My plan was to replace all the voice devices on my home with this one, but they left out the fact that you have 3 choices of wake words.  I hope this will change VERY soon.  I hate saying "Hey Jarvis tell Amie.....".

mcbutterpants007, 2025-04-22:
> Hey Rob, just wanted to say thank you! I've been searching for a way to get custom wake words since the PE showed up. I made a custom wake word and used it with atom echos and was very disappointed [...]

If possible can you link to a write up for this, or give some explanation on the process?  From what I am understanding your flashing the PE with ESP home then installing wake word voice assistants from [here](https://github.com/esphome/wake-word-voice-assistants).  Am I on the right track??

mcbutterpants007, 2025-04-28:
@RobMeades I was trying to flash the PE this morning but the code gave me this: 

Could not find __init__.py file for component aic3204. Please check the component is defined by this source (search path: /data/external_components/3bf38338/esphome/components/aic3204/__init__.py).
list: The list of components to use from the external source. By default, all available components are used.

I will hunt this down after work, but I wanted to make sure I wasn't missing anything

nickclark7, 2025-05-24:
Also disappointing to discover the home assistant voice preview edition does not support microwakeword.

After testing the M5 I found wake word processing worked much better when done locally on that hardware then it was being processed by home assistant. I found wake word processing on the home assistant voice preview edition to work about the same as the M5 using HA processing which is a bit disappointing.

gshpychka, 2025-05-26:
> Also disappointing to discover the home assistant voice preview edition does not support microwakeword

It does. That's the only wake word lib it supports.

kkuriata, 2025-05-29:
I'm surprised as well. I bought this device to use it with snowboy/openwakeword, but I can't...

Makr91, 2025-10-20:
Does anyone have any updates? or does anyone have any hacks? at this rate I am about to go back to my raspberry assistants.

neovinter, 2025-11-06:
+1 We really need this feature!