esphome/home-assistant-voice-pe #335: German Tranlation Update AN -> EIN

greyhound3, 2025-02-20:
It is really annoying the the hassturnoff event just listens to the german word "AN". The opposide of "AN" is "AB", but nobody is using this. Everybody says, turn the light "EIN"(ON), which the opposide is "AUS"(OFF). It is correctly stated by the assistent, "turned the light EIN", instead of "AN". I tried to change it in the sentences.yaml file by adding "(ein|an)", but it seems to get auto-deleted with each Voice-to-Phrase restart.
SOLUTION:
Please change it the automatic reconstruction of the yaml file  to "(ein|an)" as default.
UPDATED CONPOSED YAML (line 336 ...):
HassTurnOn:
    data:
    - sentences:
      - <schalte> <alle_lichter> an
    - sentences:
      - <schalte> <name> an
      - <schalte> <name> <area_floor> (ein|an)
      - <schalte> <area_floor> <name> (ein|an)
      requires_context:
        domain:
        - light
        - switch
        - fan
        - media_player
        - input_boolean
    - sentences:
      - <schalte> <alle_lichter> <area_floor> (ein|an)
      - <schalte> <area_floor> <alle_lichter> (ein|an)

jf-uu, 2025-02-20:
It should already support `ein` from what I can tell:

https://github.com/home-assistant/intents/blob/f6618600e85d99a621b684282466555bb3354dff/sentences/de/light_HassTurnOn.yaml#L3-L23

`<an>` expands to this:

https://github.com/home-assistant/intents/blob/f6618600e85d99a621b684282466555bb3354dff/sentences/de/_common.yaml#L544

Which already includes `ein`.

greyhound3, 2025-02-20:
I did not know, how hard it was to change an assistance, using rhasspy 2.5 -> voice-to-phrase. AN->EIN made me say everything twice, LOL

greyhound3, 2025-02-22:
I posted my solution here https://github.com/OHF-Voice/speech-to-phrase/issues/24#issuecomment-2676419419