esphome/home-assistant-voice-pe #336: Companion App Does not request Bluetooth Permission

mendezg, 2025-02-21:
Setup of the Voice Assistant PE is not possible on iOS Companion App if the Companion App has never requested access to bluetooth.

iOS apps can only be given permission to use bluetooth via the iOS security settings if the app has requested permission. If the app has not requested permission, then it will not appear in the list of apps available to grant permission. If one does not give the app permission to use bluetooth then it will be unable to connect to the VA PE and complete setup. 

I have tried with both my existing companion app that I have used with home assistant for a long time, as well as a fresh install on a new phone and neither have the ability to grant permission to the app nor could I find a way within the app to trigger the request for bluetooth access.

jaredharley, 2025-02-23:
Just now, I followed the [Getting Started steps](https://voice-pe.home-assistant.io/getting-started/) to set up my very first PE. I was a bit confused about this part of the prerequisites section, because when I followed them, I did not have a bluetooth option to enable for HomeAssistant (because the app had never requested it before):

>- Make sure the Home Assistant Companion App has Bluetooth permission:
>     - On iPhone, go to Settings > Privacy & Security and enable Bluetooth for Home Assistant.

I continued following the steps and was prompted for the Bluetooth permissions (below) only when I went to the Settings > Devices & services section of the app. I had to touch "Continue" on the first button, and then proceed past a second precise location prompt.

[image]

After passing these prompts, the bluetooth option is present in Privacy & Security > HomeAssistant. I was able to follow the rest of the onboarding steps to complete the configuration of my PE.

HomeAssistant Supervised 2025.2.4
iOS Companion App 2025.1

mendezg, 2025-02-23:
Do you know how you triggered that? I've spent hours trying various things to see if I can get the app to request bluetooth permissions without luck.

jaredharley, 2025-02-27:
It popped up on its own when I did the following on a phone I'd never set up devices with before:

In the Home Assistant app on the iPhone, touch the three lines for the menu, and touch Settings at the bottom.
Touch the "Devices and Services" option. This is where I got the Bluetooth permissions pop up I screenshotted above. 

Since seeing that prompt, I have not found a way to retrigger it or verify Bluetooth settings in the app.