esphome/home-assistant-voice-pe #337: Device stops playing stream and returns to idle at random intervals.

nevvob, 2025-02-25:
Whilst playing a radio stream through the device after a short, usually random between 5-15 minutes the music stops and the device switches back to Idle, it does this regardless of whether I trigger it from an automation, or select the radio station manually through the browser, it also does this for music being streamed from a samba share on the local network.

I have made sure it is running the latest firmware.

Please see https://community.home-assistant.io/t/voice-assistant-pe-randomly-stops-playing-radio-station/840851 for other users experiencing the same.

Happy to help debug.

Thanks

kahrendt, 2025-03-04:
Could you try using Music Assistant to send the audio instead of using the radio browser in HA and reporting back if it is still problematic?

bradwick, 2025-03-04:
I also have this issue. I exclusively play filesystem content from Music Assistant. Most recent stable versions of PE firmware, Music Assistant, and Home Assistant.

nevvob, 2025-03-05:
Hi - yes it happens if I stream files from the SSD on my Home Assistant Yellow as well.

jgdf42, 2025-04-12:
[image]

I'm having the same issue playing back locally stored mp3. I retired my google home speakers to use voice device with an external speaker and the results are worse due to this bug.

tommyjlong, 2025-04-15:
Playing from the radio to the Voice PE use to work (forget the release but around 25.1.0 when I last tested), but now it only plays a couple of seconds and fails (this is after the VAPE has been running a couple of days).

Here is the versions:
```
[17:53:00.842][I][app:100]: ESPHome version 2025.3.3 compiled on Apr  1 2025, 00:42:56
[17:53:00.842][I][app:102]: Project Nabu Casa.Home Assistant Voice PE version 25.3.4
```
Here is the error
```
[17:43:23.731][D][media_player:074]: 'Media Player' - Setting
[17:43:23.733][D][media_player:081]:   Media URL: https://MY_HA_URL/api/esphome/ffmpeg_proxy/af4c9c675d7cb094c1dac9ec15481fa6/RP8fjACrQz0R87liRNxVMA.flac
[17:43:23.734][D][speaker_media_player:426]: State changed to PLAYING
[17:43:24.298][D][esp-idf:000][med_read]: I (532236525) esp-x509-crt-bundle: Certificate validated

[17:43:24.429][D][ring_buffer:034][med_read]: Created ring buffer with size 1000000
[17:43:24.429][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[17:43:27.879][D][speaker_media_player.pipeline:124]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[17:43:37.069][D][ring_buffer:034]: Created ring buffer with size 19200
[17:43:37.070][D][speaker_mixer:310]: Starting speaker mixer
[17:43:37.083][D][speaker_mixer:318]: Started speaker mixer
[17:43:38.823][E][speaker_media_player:330]: The media pipeline's audio decoder encountered an error.
[17:43:40.043][D][speaker_media_player:426]: State changed to IDLE
[17:43:40.044][D][speaker_mixer:323]: Stopping speaker mixer
```
Here is the action I use for testing purposes:
```
action: media_player.play_media
data:
  media_content_id: >-    
    https://samcloud.spacial.com/api/listen?sid=111415&rid=194939&f=mp3,any&br=128000,any&m=sc
  media_content_type: audio/mpeg
target:
  device_id: MY_HA_VAPE_DEVICE_ID
```

If I power cycle the VAPE, it works again, but sometimes only for a couple of minutes, and then rolls over without any errors in the logs.
(BTW, I'm connecting to VAPE using aioesphomeapi-logs to capture logs)

Mirkbot, 2025-05-10:
I am experiencing the same issue. It used to work for me (unfortunately I can't tell exactly with which update it stopped), but now the voice PE goes to idle and pauses playback after a couple of seconds every time I try to play some media.

jszust, 2025-05-14:
I’m having a very similar issue. My voice PE will play the first song in a Tidal playlist then will go idle and the playlist queue clears and no other songs play

closed 2025-05-21 by jesserockz

xkjq, 2025-08-03:
Just to confirm this issue still occurs on 2025.7.0

Happy to provide logs if someone can guide me how to get them.

2600box, 2025-08-07:
Hello, I found this thread searching for the same issue, except I don't don't use voice PE or music assistant or anything like that as far as I know.

I have a dlna speaker, and have home assistant media player, play a long nature sounds file from the media folder, on single repeat. 

At some point randomly, it returns to idle. Can be an hour or 36 hours... 

I am thinking to solve this by monitoring for the idle status and restarting automatically at that time.

This would appear to be a home assistant core issue? Or a coincidence?

PR #392 (2025-05-21): Update to ESPHome 2025.5 - use upstream audio components
This will currently fail CI tests, since ESPHome 2025.5.0 hasn't been released yet.

- Updates the configuration to use the upstream audio code in ESPHome 2025.5. 
  - Microphones are handled differently - uses callbacks instead of restricting each microphone to exactly 1 downstream reader
  - Uses built-in actions to enable and disable the "stop" wake word instead of using lambdas
- Removes all external components' code (except for ``voice_kit``, which is not planned to be merged upstream).

This will finish implementing a couple outsta [... cut]