esphome/home-assistant-voice-pe #338: Response message not played on last yaml

bigboban, 2025-02-27:
I created issue on Esphome repo, but today i confirm same problem on Voice assistant PE. Same behavior, same message in log file: https://github.com/esphome/issues/issues/6807

It seems this PR https://github.com/esphome/home-assistant-voice-pe/pull/326 broke more than expected.

bigboban, 2025-02-27:
And one other note. I see this line on old/correct fw (2024.12):

`[20:11:30][D][esp-idf:000][ann_read]: I (133027) esp-x509-crt-bundle: Certificate validated`

This line is not on new fw/esphome. Probably some https/certificate validation problem.

StrandmonYellow, 2025-03-02:
I am experiencing a similar issue. When I have a custom response set in my automation, it takes a very long timd for the speaker to actually play the response. It used to be allmost instantly.

kahrendt, 2025-03-04:
> 
> This line is not on new fw/esphome. Probably some https/certificate validation problem.

How is your https certificate signed?

bigboban, 2025-03-04:
> How is your https certificate signed?

Valid lets-encrypt certificate. URL is in external form and it is fully https accessible from internet and LAN too.

kahrendt, 2025-03-04:
Are you compiling the VPE firmware yourself through the ESPHome device builder? If so, could you please try to run the stock firmware? You can reinstall the stock firmware [here](https://esphome.github.io/home-assistant-voice-pe/) (please try version 24.2.1)

kahrendt, 2025-03-04:
> But i have this: `Firmware: 25.1.1 (ESPHome 2024.12.4)`

On that page you should be able to select the custom version. Only 25.2.0 and 25.2.1 (sorry I had 24 instead of 25 in my last comment) use the new media player code, but those are beta firmware, hence not the default.

I'm wondering if it has something to do with not having the ``http_request`` component. That only exists in the factory firmware yaml, so if you build it yourself it yourself you wouldn't have it compiled in.

bigboban, 2025-03-04:
Ok, official release is ok. But i don't understand what you wrote here "That only exists in the factory firmware yaml, so if you build it yourself it yourself you wouldn't have it compiled in"

So THIS source https://github.com/esphome/home-assistant-voice-pe/blob/dev/home-assistant-voice.yaml is not source for official firmwares? So what is the point of this yaml? I clarified this in original bug report above.

So - i report this https://github.com/esphome/home-assistant-voice-pe/blob/dev/home-assistant-voice.yaml yaml is invalid and has bug reported above. Maybe you are right there is no http-request component.. but why?

kahrendt, 2025-03-04:
Thanks for the quick response! 

Sorry I was unclear by what I meant by the factory yaml! I was specifically referencing the ``home-assistant-voice.factory.yaml`` in this repo. That yaml file basically just handles setting up the updates in HA for the device (along with the Improv for configuring wifi), and it specifically adds the ``http_request`` component, which is necessary for updating in HA. That yaml then includes the ``home-assistant-voice.yaml``, which has the configuration for the main purpose of the device (e.g., voice assistant and the media player). When you take control in ESPHome device builder, you only want the second config file. Otherwise you would still get updates in HA and that would erase any custom changes if you accidentally update it.

I'm going to close this particular issue since it isn't directly due to the VPE firmware but rather a general ESPHome speaker media player bug. I'm going to follow up on your other issue when I figure out a fix for this. For the record, there are two https-related issues I'm working on
- This one/your other linked issue where it just doesn't work without the http request component: <https://github.com/esphome/issues/issues/6807>
- If you add the http request component on an ESP32, it will crash if it tries to validate the certificate (ESP32 S3's seem fine): <https://github.com/esphome/wake-word-voice-assistants/issues/71>

closed 2025-03-04 by kahrendt