esphome/home-assistant-voice-pe #341: Responses incoming on multiple devices

DevGoran, 2025-03-02:
Hi

I have 2 HA Voice Preview editions, a pipeline using llama3.1, faster-whisper and piper.

Weirdly if I ask one of the 2 HA Voices something like "What's 3 divided by 2?", it will response on the second one first something like "3 divided by 2 is 1.5" then right after it responds to the Voice I initially talked to with something like "1.5" or "I already told you, it's 1.5".

Also noticed that one of the 2 Voices also responds if I use Assist on the phone. So, if I talk to the one in Room 1, first I will hear the response in room 2 and then room 1. If I talk to the one in room 2, the one in room 1 will first respond and then the one in room 1. If I talk to assist on the phone, the one in room 1 will respond and afterwards I will hear a response on the phone. I think this issue started after adding the second Voice Preview Edition to my home.

kahrendt, 2025-03-04:
What happens if you disable the LLM and just use the basic assist commands? My guess is the LLM is using the broadcast intent to play the messages on the other devices. This is a harder thing to fix, because what the LLM does is all a blackbox. If this is the source of the issue, you could try a different LLM instead of llama3.1.

DevGoran, 2025-03-04:
Llama 3.1:8b seems like to be it. Llama 3.2:3b does not cause the issue.

kahrendt, 2025-03-04:
Thanks for reporting!

closed 2025-03-04 by kahrendt