esphome/home-assistant-voice-pe #344: Communication with Voice Kit failed

nrjize, 2025-03-03:
Happened around a week ago after an update, however my Voice PE was bricked and no longer communicating with microphone or speaker

From log:
`[14:33:38][E][voice_kit:026]: Communication with Voice Kit failed
[14:33:38][E][component:119]: Component voice_kit was marked as failed.
[14:33:38][E][component:164]: Component voice_kit set Error flag: unspecified

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/19060970/esp-web-tools-logs.txt)

`

kahrendt, 2025-03-17:
This seems to be a hardware failure where the XMOS chip is failing to communicate with the ESP32 chip. Do you know if the failure happened immediately after an update or just around the same time?

nrjize, 2025-03-17:
I received prompts for a updates, not sure if it happened right after one as I wasn't checking functionality right after applying them, unfortunately.

kahrendt, 2025-03-17:
> I received prompts for a few updates, not sure if it happened right after one as I wasn't checking functionality right after applying them, unfortunately. I guess I assumed it was due to an update as opposed to a hardware failure

Thanks for reporting! If you haven't done so already, please contact your distributor to return and get a replacement due to the hardware failure.

closed 2025-03-17 by kahrendt