esphome/home-assistant-voice-pe #349: Support USB provisioning

DemiMarie, 2025-03-09:
Instead of requiring the use of the Home Assistant mobile app, it would be more convenient to be able to connect the Voice PE to the Home Assistant over USB and have provisioning occur automatically.

MichaelMKKelly, 2025-03-11:
you can do this if you install firmware using the web installer then you can insert wifi details as part of the process.
https://voice-pe.home-assistant.io/guides/firmware-reinstall/

DemiMarie, 2025-03-13:
Ideally, this could be done from Home Assistant itself, preferably server-side (by plugging the Voice PE directly into Home Assistant).

MichaelMKKelly, 2025-03-13:
you could do that with the esphome builder addon if you wanted to. by creating a device, adding the yaml found in this repo. then modify to include your wifi credentials. then you can build the firmware and install it via a usb port on your HA server.

DemiMarie, 2025-03-13:
I think it would make sense to include this as an in-box feature.

Also, the data that needs to be provisioned includes not only the Wi-Fi password, but also a Noise pre-shared key.