esphome/home-assistant-voice-pe #351: [REQUEST] Extend and improve ESPHome's native media player(s) to make it/them have both high-quality audio output and advanced music playback features similar to Squeezelight client?

Hedda, 2025-03-11:
Does Nabu Casa developers have a plan to further improve ESPHome's native media player(s) for music playback?

Are there any ongoing development efforts or roadmap to make ESPHome native media player have much more advanced music playback features like multi-room audio grouping and stereo pair support similar to Squeezelight client to be able to make Home Assistant Voice Preview Edition hardware and similar products based on ESPHome and ESP32 + XMOS act like a reference that is capable of replacing other so-called "music streamers" (i.e. network music streamer receivers meant for)?

Could there be a roadmap goal of improving ESPHome native media player(s) to make it have feature parity with competing open-source music streaming clients like Squeezelite and Snapcast or similar?

PS: My reason to hope and push for such efforts are partially because all of Google's Chromecast Audio (audio-only Google Cast device) dongle just now stopped working globally for everyone in the whole world as there are no alternatives which are inexpensive and have similal feature set, like multi-room audio grouping and stereo pair support. So for exsiting Chromecast Audio users there is currently no equivalent Google Cast audio-only product you can buy as a replacement as a Google Cast audio (music) music receiver and network-attached audio-jack output adapter product, so many users are stuck in limbo now. For more information see this Home Assistant community forum thread: 

* https://community.home-assistant.io/t/chromecast-audio-and-2nd-gen-chromecast-video-all-stopped-working-worldwide-in-google-home-with-untrusted-device-error/861989

janhense, 2025-03-11:
Maybe you can use the Music Assistant Addon https://www.home-assistant.io/blog/2024/05/09/music-assistant-2/ . There is also a blueprint for voice control [https://github.com/music-assistant/voice-support]

Hedda, 2025-03-12:
> Maybe you can use the Music Assistant Addon https://www.home-assistant.io/blog/2024/05/09/music-assistant-2/ .

Not by itself as Music Assistant depends on what media player features are supported by the "Player Provider" on the client-side, see:

* https://www.music-assistant.io/player-support/

As the fact is ESPHome's native media player(s) does not yet support those features (like audio grouping and stereo pair, etc).. See:

[image]

Hedda, 2025-03-18:
> I spoke to the music assistant team and they mentioned they do have plans to get voice PE working as a multi-room setup with snapcast but I don't think there is a timeline.
> 
> You can see it here https://github.com/orgs/music-assistant/discussions/3391

Yes have seen that, though again noted that support for Squeezelite client/  Slim Protocol (a.k.a. SlimProto) is currently the best today:

* https://www.music-assistant.io/player-support/squeezelite/

Subsonic/Airsonic could be another open-source alternative today, but Music Assistant do not yet have an Subsonic player provider:

* https://www.music-assistant.io/player-support/

PS: Long-term for the future I hope that "Matter Casting" also will be extended upstream too and adopted by the whole industry:

* https://github.com/esphome/home-assistant-voice-pe/issues/353

kahrendt, 2025-05-22:
This is a more general ESPHome feature request rather than a VPE specific request, so I'm going to close it here.

We are working on adding synced audio support to ESPHome, but it is in the early stages. One task related to it as already available on our project board: https://github.com/orgs/esphome/projects/7?pane=issue&itemId=109541647&issue=esphome%7Cbacklog%7C20

closed 2025-05-22 by kahrendt