esphome/home-assistant-voice-pe #354: Home Assistant Voice PE microphone not responding to wake word activation; no sound through media player

bwip, 2025-03-14:
Similar to #321

Successfully added the device to Home Assistant.
It worked for one day.

After one day, no response on the wake word.

Verified that the device appears in the Home Assistant UI.
Attempted multiple wake words with no response.
Repeated the setup process multiple times (factory reset) without success.

The device is controlable via HA (led ring, ...).

The media player and voice inputs do not work and produce a lot of errors in the esp log:

```
[23:17:25][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:25][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:25][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:25][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][D][voice_assistant:646]: Event Type: 0
[23:17:26][E][voice_assistant:783]: Error: stt-stream-failed - speech-to-text failed
[23:17:26][D][voice_assistant:628]: Signaling stop...
[23:17:26][D][voice_assistant:509]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[23:17:26][D][voice_assistant:516]: Desired state set to IDLE
[23:17:26][D][voice_assistant:509]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[23:17:26][D][light:036]: 'voice_assistant_leds' Setting:
[23:17:26][D][light:051]:   Brightness: 76%
[23:17:26][D][light:058]:   Red: 100%, Green: 0%, Blue: 0%
[23:17:26][D][light:109]:   Effect: 'Error'
[23:17:26][D][voice_assistant:646]: Event Type: 2
[23:17:26][D][voice_assistant:742]: Assist Pipeline ended
[23:17:26][D][voice_assistant:509]: State changed from STOPPING_MICROPHONE to IDLE
[23:17:26][D][voice_assistant:509]: State changed from IDLE to IDLE
[23:17:26][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][D][power_supply:033]: Enabling power supply.
[23:17:26][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_SIZE
[23:17:26][D][power_supply:033]: Enabling power supply.
[23:17:27][D][power_supply:033]: Enabling power supply.
[23:17:27][D][light:036]: 'voice_assistant_leds' Setting:
[23:17:27][D][light:047]:   State: OFF
[23:17:27][D][light:109]:   Effect: 'None'
[23:17:37][D][power_supply:048]: Disabling power supply.
```

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/19255388/esp-web-tools-logs.txt)

Is this a hardware problem?

BR

kahrendt, 2025-03-17:
Could you capture and post the logs right after rebooting showing all the setup? I'm suspecting a hardware failure where the XMOS chip isn't running and so it isn't generating a clock signal and the I2S driver can't send or receive audio data.

bwip, 2025-03-17:
Thank you for your interest, here is a log with a reboot: [esp-web-tools-logs (1).txt](https://github.com/user-attachments/files/19299229/esp-web-tools-logs.1.txt)

I returned the unit today and wait for a new one. The short time I could test it I was happy with it, especially with the good wake word detection.

kahrendt, 2025-03-17:
> Thank you for your interest, here is a log with a reboot: [esp-web-tools-logs (1).txt](https://github.com/user-attachments/files/19299229/esp-web-tools-logs.1.txt)
>
> I returned the unit today and [...]

Thanks for those logs, they confirm a hardware failure with the XMOS chip since it says the voice_kit component failed. I'm glad to hear you returned it and requested a new one, as that should fix the problem.

closed 2025-03-17 by kahrendt

ahmedsamer, 2025-04-18:
I'm facing the same problem, should I return it as well?