esphome/home-assistant-voice-pe #355: TTS fails for longer responses (CancelledError + ESP_FAIL)

hheimbuerger, 2025-03-19:
I just got my first Voice PE and set it up. STT using a wyoming-whisper Docker container is working fine. The conversation agent is working fine.

TTS using a wyoming-piper Docker container only works for short responses. On anything longer than two sentences, the Voice PE gives me the counterclockwise spin for ~2s, then it just turns off, no response.

In the voice assistant debug view, it looks like this:

[image]

The HA log doesn't have a helpful error message that would indicate any specific troubleshooting steps:

```
2025-03-19 11:20:54.602 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback SpeechManager._async_get_tts_audio.<locals>.handle_error() at /usr/src/homeassistant/homeassistant/components/tts/__init__.py:844 (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/tts/__init__.py", line 846, in handle_error
    if audio_task.exception():
       ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/components/tts/__init__.py", line 786, in get_tts_data
    extension, data = await engine_instance.internal_async_get_tts_audio(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        message, language, options
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/components/tts/__init__.py", line 492, in internal_async_get_tts_audio
    return await self.async_get_tts_audio(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        message=message, language=language, options=options
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/components/wyoming/tts.py", line 126, in async_get_tts_audio
    event = await client.read_event()
            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/wyoming/client.py", line 25, in read_event
    return await async_read_event(self._reader)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/wyoming/event.py", line 79, in async_read_event
    json_line = await reader.readline()
                ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 562, in readline
    line = await self.readuntil(sep)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 677, in readuntil
    await self._wait_for_data('readuntil')
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 539, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError
2025-03-19 11:20:54.616 ERROR (MainThread) [homeassistant.components.esphome.manager] Home Assistant Voice Preview: [E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
```

This is also reproducible when switching the TTS component to Google Translate.

In the Piper container, it looks like this:

```
ERROR:asyncio:Task exception was never retrieved
future: <Task finished name='wyoming event handler' coro=<AsyncEventHandler.run() done, defined at /usr/local/lib/python3.9/dist-packages/wyoming/server.py:31> exception=BrokenPipeError(32, 'Broken pipe')>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/wyoming_piper/handler.py", line 48, in handle_event
    return await self._handle_event(event)
  File "/usr/local/lib/python3.9/dist-packages/wyoming_piper/handler.py", line 132, in _handle_event
    await self.write_event(
  File "/usr/local/lib/python3.9/dist-packages/wyoming/server.py", line 29, in write_event
    await async_write_event(event, self.writer)
  File "/usr/local/lib/python3.9/dist-packages/wyoming/event.py", line 131, in async_write_event
    await writer.drain()
  File "/usr/lib/python3.9/asyncio/streams.py", line 387, in drain
    await self._protocol._drain_helper()
  File "/usr/lib/python3.9/asyncio/streams.py", line 190, in _drain_helper
    raise ConnectionResetError('Connection lost')
ConnectionResetError: Connection lost

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/wyoming/server.py", line 41, in run
    if not (await self.handle_event(event)):
  File "/usr/local/lib/python3.9/dist-packages/wyoming_piper/handler.py", line 50, in handle_event
    await self.write_event(
  File "/usr/local/lib/python3.9/dist-packages/wyoming/server.py", line 29, in write_event
    await async_write_event(event, self.writer)
  File "/usr/local/lib/python3.9/dist-packages/wyoming/event.py", line 131, in async_write_event
    await writer.drain()
  File "/usr/lib/python3.9/asyncio/streams.py", line 375, in drain
    raise exc
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 918, in write
    n = self._sock.send(data)
BrokenPipeError: [Errno 32] Broken pipe
```

Possibly related issues:
* https://github.com/home-assistant/addons/issues/3360: but it's a) for the addon, b) marked as 'not planned', whatever that means, and c) doesn't mention the `ESP_FAIL`
* https://community.home-assistant.io/t/http-timeout-for-voice-assistant-pe-even-though-the-response-is-recieved/834200: but they mention that switching to another TTS fixes the issue, which isn't the case for me

slazix54, 2025-03-20:
same man, it's been killing me. I don't get the connection lost errors I just get this mess. I can still get responses for stuff that can be explained in 1-2 sentences, but everything else is hooped af.

` File "/usr/src/homeassistant/homeassistant/components/tts/__init__.py", line 786, in get_tts_data
    extension, data = await engine_instance.internal_async_get_tts_audio(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        message, language, options
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/components/tts/__init__.py", line 492, in internal_async_get_tts_audio
    return await self.async_get_tts_audio(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        message=message, language=language, options=options
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/openai_tts/tts.py", line 247, in async_get_tts_audio
    return await asyncio.shield(
           ^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
asyncio.exceptions.CancelledError`

yorb, 2025-03-27:
Same issue here. It's frustrating to see all these closed issues. I've seen a couple workarounds but they seem like too much work to fix what seems like basic functionality. Makes me think it must be something special with our setups, but mine is as basic as they come: rpi4 + haos.

codesalatdev, 2025-03-27:
@yorb did you see the merged PR that mentions this issue above your comment? The update isn't out yet, but it seems to be fixed in the dev branch according to the description

yorb, 2025-03-27:
> @yorb did you see the merged PR that mentions this issue above your comment? The update isn't out yet, but it seems to be fixed in the dev branch according to the description (or at least "partially address" it :D) 

Not sure how I missed that, nice!

execdd17, 2025-03-29:
Hey all, I installed the firmware from the PR you mentioned (#356) but I'm still experiencing what I think is the same issue. I'm running [this code](https://github.com/nalf3in/wyoming-glados/blob/main/server/handler.py) for TTS via docker on my desktop's GPU. 

I'm trying to wrap my head around the issue, is the wav [chunk size too big](https://github.com/nalf3in/wyoming-glados/blob/main/server/handler.py#L112), or some other kind of overloading of the device? 

Thanks

codesalatdev, 2025-03-29:
> I'm trying to wrap my head around the issue, is the wav [chunk size too big](https://github.com/nalf3in/wyoming-glados/blob/main/server/handler.py#L112), or some other kind of overloading of the device?

It seems to be way easier than that: ESPHome has a timeout limit until it has to receive initial audio data from the server (time between HTTP Headers and actual data in the body). That is why you only have the issue with the Voice PE but not with your phone or other clients.

The PR that mentioned this issue references ESPHome v2025.2.2, which changes 1) how the timeout is calculated and 2) bumped the timeout to 5000ms (https://github.com/esphome/esphome/pull/8341). If you're still hitting this limit - be it because Piper takes some time to generate the audio responses (because it doesn't "stream" the audio yet) or some other issue - you will still get the same issue.

StrandmonYellow, 2025-04-02:
It has been an issue from the beginning when using the Voice PE. And as far as I can see, text-to-speech always takes 0s according to the HA logs. I would be nice if this got fixed. When I am asking a question to an LLM, I don't mind to have to wait for a bit because I am already expecting it to take a second for the LLM to generate an answer.

hheimbuerger, 2025-04-02:
> It has been an issue from the beginning when using the Voice PE. And as far as I can see, text-to-speech always takes 0s according to the HA logs.

That's not correct! This screen definitely reports correct generation time whenever it actually generates a response, i.e. when there's only 1-2 sentences of output from the conversation agent.

execdd17, 2025-04-02:
Hey everyone! I have some updates with my TTS issues. What remained the same is that my STT, LLM, and TTS always run on an RTX 3090 desktop machine. What changed:

1. I upgraded my home assistant device from a Raspberry Pi 3B+ to a Pi 5 (8GB)
2. I updated the voice assistant pe to Firmware: 25.3.4 (ESPHome 2025.3.3)

The TTS exceptions have dramatically reduced for both long and short outputs. I observed only one instance where the voice assistant pe seems to have crashed and rebooted midway through playing the TTS output. I'm also seeing rare instances of exceptions within the TTS docker container (which is new). Overall though, there has been a massive improvement and the interactive experience is much better. 🎉 

Also fwiw, **anecdotally**, the TTS processing time reported by HA is always 0 or near 0. I assumed this was because the audio bytes are processed as a stream and started quickly. I'm curious if anyone knows more about that!

hheimbuerger, 2025-04-02:
I have to correct myself, you're right that it shows `0s` for Piper responses, even if they are very short.

That's not the case for Google Translate as the TTS, though:
[image]
Maybe that's what I remembered.

hheimbuerger, 2025-04-02:
I've also installed the firmware 25.3.4 (based on ESPHome 2025.3.3) now.

I'm now able to generate longer responses using Google Translate TTS (weirdly enough, it still shows `0s` in these cases), but nothing seems to have improved for the local Piper container:

Short (single sentence) responses are still generating perfectly fine (except showing `0s` processing time), while long responses generate no output. Still seeing a `CancelledError` and the following in the logs:

```
[E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[E][speaker_media_player:334]: The announcement pipeline's file reader encountered an error.
[W][micro_wake_word:375]: Wake word dection is already running
[E][speaker_media_player:339]: The announcement pipeline's file reader encountered an error.
```

Looks like I have now regressed (or progressed?) to the state of the person with the [aforementioned forum topic](https://community.home-assistant.io/t/http-timeout-for-voice-assistant-pe-even-though-the-response-is-recieved/834200) (different error messages, though), for whom switching to another TTS solution resolved the issue.

gshpychka, 2025-04-06:
I'm experiencing a different faliure mode - no erros anywhere I can find, and the audio plays just fine in the debug view. 

But it still just silently stops with long respones.

User873902, 2025-05-08:
> 2\. I updated the voice assistant pe to Firmware: 25.3.4 (ESPHome 2025.3.3)

I'll give it a shot and report back. Enabled beta.

einord, 2025-06-15:
Still an issue in 2025.6.1

nconder, 2025-06-18:
2025.6.1 core 25.05.5 supervisor - playback still stops at about 10 seconds.

hheimbuerger, 2025-11-28:
> **Why this happens:**
> 
> * Voice PE tries to download TTS audio from HA via HTTP
> * If HA provides Docker-internal IP, Voice PE can't reach it

@thommaehh81 If Voice PE couldn't reach HA at all, then how did short messages (<10s) get played back correctly in your case!?

I cannot imagine that they did, so therefore it seems you had a different root cause to what's documented here. Which also means your solution doesn't fit the described issue.