esphome/home-assistant-voice-pe #357: Repeat Mode on Voice Assistant PE and breaking the timer.

ilar, 2025-03-21:
It appears that the timer functionality on the Voice Assistant PE relies on the music repeat mode being set to 'one', which can break if you ever change that value.

At the moment, if you use your VA PE to play music (via external speakers, presumably), you will pretty much HAVE that turned to 'false' instead. This breaks the timer- it will ring once, and then not again, though the light on it will continue to flash until you catch its attention via wake word.

I'm of the opinion that the voice assistant PE should probably cache the current repeat mode setting, change to repeat mode 'one', play the timer, and then when the timer is ended, restore the previous value. This would cause the timer- and any files you play over the VA PE- to behave as expected.

kahrendt, 2025-03-28:
The media player essentially two different players in one (pipelines). One handles regular media like music, and the other handles announcements and sound effects on the device. When the timer needs to start beeping, the firmware turns on the repeat mode only for the announcements. When the timer stops beeping, it disables the repeat mode for the announcements. The media pipeline will still continue to play without repeating throughout this process, as its repeat mode state is independent.

Currently, it is only possible to change the repeat mode in actions specified by the actual yaml config, you can't do that in HA. So the only time the repeat mode changes is in the process I outlined above.

Have you encountered things repeating when they shouldn't on the basic firmware?

ilar, 2025-03-30:
The problem I'm running in to is that on basic firmware, the alarm _won't_ repeat when it should be- it'll ring once, then fall silent even while the light ring still flashes for the alarm.

kahrendt, 2025-03-31:
I think I've managed to reproduce the same (or a closely related) bug. If I have music playing, then pause it, and then have the timer fire, it will only play the sound once. Are you able to have the timer only ring once in any under circumstance?

PR #372 (2025-04-01): Bump ESPHome to v2025.3.3
Includes several fixes related to an incorrect pause state:
- esphome/esphome#8474 - correctly change from pause to idle state after receiving a stop command
  - fixes https://github.com/music-assistant/support/issues/3775 
- esphome/esphome#8488 - announcement playlist doesn't follow playlist and change to the correct state after starting a new file while paused
  - fixes #357 
  - fixes #370
closed 2025-03-31 by kbx81