esphome/home-assistant-voice-pe #368: 2025.3.2 Beta - Bell ringing

Chompy80, 2025-03-28:
Feature request - Please make the bell played to acknowledge requests or preempt announcements in 2025.3.2 beta an optional feature.

kahrendt, 2025-04-02:
This [HA PR](https://github.com/home-assistant/core/pull/141930) makes it easier to disable the default bell ringing sound (before you had to specify the pre-announce media id as null).

closed 2025-04-02 by kahrendt

MorganBridget, 2025-04-03:
> This [HA PR](https://github.com/home-assistant/core/pull/141930) makes it easier to disable the default bell ringing sound (before you had to specify the pre-announce media id as null).

How do I install this?

kahrendt, 2025-04-03:
> > This [HA PR](https://github.com/home-assistant/core/pull/141930) makes it easier to disable the default bell ringing sound (before you had to specify the pre-announce media id as null).
> 
> How do I install this?

It's part of Home Assistant release 2025.4. You don't need to do anything extra besides update your HA instance.

MorganBridget, 2025-04-03:
> Feature request - Please make the bell played to acknowledge requests or preempt announcements in 2025.3.2 beta an optional feature.  I would prefer to have it turned off and only use media files when they make sense for my use case.

How to install the fix and disable this annoying bell?

MorganBridget, 2025-04-03:
> It's part of Home Assistant release 2025.4. You don't need to do anything extra besides update your HA instance.

Great. I have that update but I still hear the ding. I feel stupid but I'm not sure where to find the setting to turn it off.

kahrendt, 2025-04-03:
You need to adjust your assist_satellite announce action. In HA, if you go developer tools and then actions and choose the assist satellite announce action, you'll see there is now a toggle switch for the preannounce sound. Disable the toggle, and then you won't have the ding anymore.

MorganBridget, 2025-04-03:
> You need to adjust your assist_satellite announce action. In HA, if you go developer tools and then actions and choose the assist satellite announce action, you'll see there is now a toggle switch for the preannounce sound. Disable the toggle, and then you won't have the ding anymore.

Oh wow. Yes I see it there. And I also see that option in all my existing automations. Is what you described the way to disable that globally? Or do I need to go into each automation and change it there? Thanks so much for your help. I'm sorry if I'm a pain.

kahrendt, 2025-04-03:
No worries! Unfortunately, you'll have to adjust each automation.

MorganBridget, 2025-04-03:
> No worries! Unfortunately, you'll have to adjust each automation.

The other problem is the preannounce now occurs when I simply type "announce testing" into the home assistant chat. Sometimes I like to play around and just type announce followed by what I want the voice pe to say and now it does that annoying "ding". I'm wondering how to disable that.

Chompy80, 2025-04-03:
> The other problem is the preannounce now occurs when I simply type "announce testing" into the home assistant chat. Sometimes I like to play around and just type announce followed by what I want the voice pe to say and now it does that annoying "ding". I'm wondering how to disable that.

The announce and start conversation actions both have the toggle for disabling the default bell. You will need to disable it in all actions.

MorganBridget, 2025-04-03:
> The announce and start conversation actions both have the toggle for disabling the default bell. You will need to disable it in all actions.

I'm not sure where to do that. I'm talking about when you use the assist text chat feature built into home assistant. Before this change if I typed announce testing. The voice pe would simply announce testing. Now it first does this ding. I am not sure where or how to turn it off when using the assist chat window.