esphome/home-assistant-voice-pe #370: Single-Press does not stop current media playback anymore

mib1185, 2025-03-29:
Hi,

I've recently updated my VPEs to 25.3.3 (_beta_) and noticed, that a single-button-press doesn't stop the current playing media anymore - I was used to stop any media playback by a single-press. Maybe it is also related to the current ha core beta (_i'm on 2025.4.0b8_)

mib1185, 2025-03-29:
**UPDATE (_tested with 25.3.3_)**

it seems to work only once after restart of the VPE

### start playing media via action call

action call
```yaml
action: media_player.play_media
data:
  media_content_id: https://mdr-284320-0.sslcast.mdr.de/mdr/284320/0/aac/high/stream.aac
  media_content_type: music
target:
  device_id: "{{ device }}"
```

vpe log:
```
[14:02:39][D][speaker_media_player:391]: State changed to ANNOUNCING
[14:02:39][D][media_player:074]: 'Media Player' - Setting
[14:02:39][D][media_player:081]:   Media URL: http://192.168.1.5:8123/api/esphome/ffmpeg_proxy/e4b1ddcbace4fffc121f3540c2b09f82/dRR1iZUSMCmEz-A7vXE-9w.flac
[14:02:39][D][media_player:087]:  Announcement: no
[14:02:39][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[14:02:39][D][ring_buffer:034][ann_read]: Created ring buffer with size 1000000
[14:02:39][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[14:02:39][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[14:02:39][D][ring_buffer:034]: Created ring buffer with size 9600
[14:02:39][D][speaker_mixer:310]: Starting speaker mixer
[14:02:39][D][speaker_mixer:318]: Started speaker mixer
[14:02:40][D][speaker_media_player.pipeline:124]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[14:02:40][D][speaker_media_player:391]: State changed to PLAYING
[14:02:40][D][ring_buffer:034]: Created ring buffer with size 19200
[14:02:48][D][power_supply:048]: Disabling power supply.
```

### single-press button

vpe log:
```
[14:03:29][D][binary_sensor:036]: 'center_button': Sending state ON
[14:03:29][D][light:036]: 'voice_assistant_leds' Setting:
[14:03:29][D][light:047]:   State: ON
[14:03:29][D][light:051]:   Brightness: 76%
[14:03:29][D][light:109]:   Effect: 'Center Button Touched'
[14:03:29][D][light:036]: 'voice_assistant_leds' Setting:
[14:03:29][D][light:051]:   Brightness: 76%
[14:03:29][D][light:055]:   Color brightness: 100%
[14:03:29][D][light:058]:   Red: 10%, Green: 77%, Blue: 100%
[14:03:29][D][power_supply:033]: Enabling power supply.
[14:03:29][D][binary_sensor:036]: 'center_button': Sending state OFF
[14:03:29][D][light:036]: 'voice_assistant_leds' Setting:
[14:03:29][D][light:047]:   State: OFF
[14:03:29][D][light:109]:   Effect: 'None'
[14:03:29][D][media_player:074]: 'Media Player' - Setting
[14:03:29][D][media_player:078]:   Command: PAUSE
[14:03:29][D][media_player:087]:  Announcement: no
[14:03:29][D][speaker_media_player:391]: State changed to PAUSED
[14:03:39][D][power_supply:048]: Disabling power supply.
```

### start playing media via action call again

interesting fact is, that the media player does `State changed to PAUSED` on the vpe

action call
```yaml
action: media_player.play_media
data:
  media_content_id: https://mdr-284320-0.sslcast.mdr.de/mdr/284320/0/aac/high/stream.aac
  media_content_type: music
[... 41 lines cut ...]
[14:05:07][D][ring_buffer:034][ann_read]: Created ring buffer with size 1000000
[14:05:07][D][speaker_media_player:391]: State changed to ANNOUNCING
[14:05:07][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[14:05:07][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[14:05:07][D][ring_buffer:034]: Created ring buffer with size 9600
[14:05:07][D][voice_assistant:509]: State changed from IDLE to START_MICROPHONE
[14:05:07][D][voice_assistant:516]: Desired state set to START_PIPELINE
[14:05:07][D][voice_assistant:222]: Starting Microphone
[14:05:07][D][ring_buffer:034]: Created ring buffer with size 16384
[14:05:07][D][voice_assistant:509]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[14:05:07][D][voice_assistant:509]: State changed from STARTING_MICROPHONE to START_PIPELINE
[14:05:07][D][voice_assistant:277]: Requesting start...
[14:05:07][D][voice_assistant:509]: State changed from START_PIPELINE to STARTING_PIPELINE
[14:05:07][D][voice_assistant:531]: Client started, streaming microphone
[14:05:07][D][voice_assistant:509]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[14:05:07][D][voice_assistant:516]: Desired state set to STREAMING_MICROPHONE
[14:05:07][D][voice_assistant:646]: Event Type: 1
[14:05:07][D][voice_assistant:649]: Assist Pipeline running
[14:05:07][D][voice_assistant:646]: Event Type: 3
[14:05:08][D][voice_assistant:660]: STT started
[14:05:08][D][light:036]: 'voice_assistant_leds' Setting:
[14:05:08][D][light:047]:   State: ON
[14:05:08][D][light:051]:   Brightness: 66%
[14:05:08][D][light:109]:   Effect: 'Waiting for Command'
```

kahrendt, 2025-03-31:
Thanks for the careful debugging! This should be fixed in the ESPHome PR just linked. Of course, it won't actually be fixed in the VPE firmware until we build with a new ESPHome release, but that is incoming!

mib1185, 2025-03-31:
thx for the fix @kahrendt 👍 just give me a sign, when the fix get available in the beta firmware and i'll test again

kahrendt, 2025-04-01:
> thx for the fix [@kahrendt](https://github.com/kahrendt) 👍 just give me a sign, when the fix get available in the beta firmware and i'll test again

25.3.4 was just pushed out with the fix, so you should get the update notification within a few hours (or you can force update to the newest version using the HA update action).

mib1185, 2025-04-02:
I can confirm, that this is solved with 25.3.4 - just for reference, following the log from the VPE

**start playing with `media_player.play_media` action**
```
[20:49:48][D][media_player:074]: 'Media Player' - Setting
[20:49:48][D][media_player:081]:   Media URL: http://192.168.1.5:8123/api/esphome/ffmpeg_proxy/e4b1ddcbace4fffc121f3540c2b09f82/T3vlUzeCJ88fOLMsL50SGw.flac
[20:49:48][D][media_player:087]:  Announcement: no
[20:49:48][D][speaker_media_player:426]: State changed to PLAYING
[20:49:48][D][ring_buffer:034][med_read]: Created ring buffer with size 1000000
[20:49:48][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[20:49:49][D][speaker_media_player.pipeline:124]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[20:49:49][D][ring_buffer:034]: Created ring buffer with size 19200
[20:49:49][D][speaker_mixer:310]: Starting speaker mixer
[20:49:49][D][speaker_mixer:318]: Started speaker mixer
```

**single-press the button**
```
[20:49:58][D][binary_sensor:036]: 'center_button': Sending state ON
[20:49:58][D][light:036]: 'voice_assistant_leds' Setting:
[20:49:58][D][light:047]:   State: ON
[20:49:58][D][light:051]:   Brightness: 76%
[20:49:58][D][light:109]:   Effect: 'Center Button Touched'
[20:49:58][D][light:036]: 'voice_assistant_leds' Setting:
[20:49:58][D][light:051]:   Brightness: 76%
[20:49:58][D][light:055]:   Color brightness: 100%
[20:49:58][D][light:058]:   Red: 10%, Green: 77%, Blue: 100%
[20:49:58][D][power_supply:033]: Enabling power supply.
[20:49:58][D][binary_sensor:036]: 'center_button': Sending state OFF
[20:49:58][D][light:036]: 'voice_assistant_leds' Setting:
[20:49:58][D][light:047]:   State: OFF
[20:49:58][D][light:109]:   Effect: 'None'
[20:49:58][D][media_player:074]: 'Media Player' - Setting
[20:49:58][D][media_player:078]:   Command: PAUSE
[20:49:58][D][media_player:087]:  Announcement: no
[20:49:58][D][speaker_media_player:426]: State changed to PAUSED
[20:50:08][D][power_supply:048]: Disabling power supply.
```

**re-start playing with `media_player.play_media` action**
```
[20:50:12][D][media_player:074]: 'Media Player' - Setting
[20:50:12][D][media_player:081]:   Media URL: http://192.168.1.5:8123/api/esphome/ffmpeg_proxy/e4b1ddcbace4fffc121f3540c2b09f82/1pmtGPkaUeNM5UFia70aTA.flac
[20:50:12][D][media_player:087]:  Announcement: no
[20:50:12][D][speaker_mixer:323]: Stopping speaker mixer
[20:50:12][D][speaker_media_player:426]: State changed to PLAYING
[20:50:12][D][ring_buffer:034][med_read]: Created ring buffer with size 1000000
[20:50:12][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[20:50:13][D][speaker_media_player.pipeline:124]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[20:50:14][D][ring_buffer:034]: Created ring buffer with size 19200
[20:50:14][D][speaker_mixer:310]: Starting speaker mixer
[20:50:14][D][speaker_mixer:318]: Started speaker mixer
```

**single-press the button**
```
[20:50:35][D][binary_sensor:036]: 'center_button': Sending state ON
[20:50:35][D][light:036]: 'voice_assistant_leds' Setting:
[20:50:35][D][light:047]:   State: ON
[20:50:35][D][light:051]:   Brightness: 76%
[20:50:35][D][light:109]:   Effect: 'Center Button Touched'
[20:50:35][D][light:036]: 'voice_assistant_leds' Setting:
[20:50:35][D][light:051]:   Brightness: 76%
[20:50:35][D][light:055]:   Color brightness: 100%
[20:50:35][D][light:058]:   Red: 10%, Green: 77%, Blue: 100%
[20:50:35][D][power_supply:033]: Enabling power supply.
[20:50:35][D][binary_sensor:036]: 'center_button': Sending state OFF
[20:50:35][D][light:036]: 'voice_assistant_leds' Setting:
[20:50:35][D][light:047]:   State: OFF
[20:50:35][D][light:109]:   Effect: 'None'
[20:50:35][D][media_player:074]: 'Media Player' - Setting
[20:50:35][D][media_player:078]:   Command: PAUSE
[20:50:35][D][media_player:087]:  Announcement: no
[20:50:35][D][speaker_media_player:426]: State changed to PAUSED
[20:50:45][D][power_supply:048]: Disabling power supply.
```

PR #372 (2025-04-01): Bump ESPHome to v2025.3.3
Includes several fixes related to an incorrect pause state:
- esphome/esphome#8474 - correctly change from pause to idle state after receiving a stop command
  - fixes https://github.com/music-assistant/support/issues/3775 
- esphome/esphome#8488 - announcement playlist doesn't follow playlist and change to the correct state after starting a new file while paused
  - fixes #357 
  - fixes #370