esphome/home-assistant-voice-pe #378: No Audio Playback for responses

hagak, 2025-04-05:
Prior to HA 2025.4 this worked fine.  However after updating my Voice Assist PE will trigger on the wake word fine and process the request as seen in the HA Voice Debug, however it does not playback the response even though Debug shows the response and can play it within HA.

If I go to the Voice Assist PE and playback media even using Piper for the media source it plays back the audio fine.  So this does not appear to be the normal issue of Local Network.  In fact if I purposely "break" my local network ip and try the playback media test it fails as expected.

AlanMW, 2025-04-06:
I am having the same issue. It was working before I updated to 2025.4, debug for the Text-to-speech shows 0s of processing which isn't realistic. I can play the audio from my computer no problem.

hagak, 2025-04-06:
Yep same here 0s of processing as well

mikey60, 2025-04-06:
My TTS processing has shown 0s for a long time even before updating to HA 2025.4.  I have never figured it out.

hagak, 2025-04-07:
Hmm not sure that is my issue since I had already done that step as noted above.  I even verified that if I set my local url incorrectly that it breaks my system even more.  IE.  set correctly I play audio via the media interface on the VA PE but if I set it incorrectly I can not.

hagak, 2025-04-07:
> I found the linked reply below and that solved my issue. In summary, I updated my local network address and my VA PE is able to reply now. Maybe it is because my TTS server is separate from my HA server.
> 
> [esphome/wake-word-voice-assistants#72 (comment)](https://github.com/esphome/wake-word-voice-assistants/issues/72#issuecomment-2679323733)

My TTS server is separate too, but you did not put the ip of the TTS server in HA local url did you?

hagak, 2025-04-07:
I am wondering if something changed on how the VA PE connects to the TTS engine in the latest HA version.  If it is being redirected to connect direct to the configured PIPER in HA then that could be my issue.  Since PIPER is in the same k8s cluster as HA I configure it to use the cluster-ip and not the external-ip.

hagak, 2025-04-07:
Adding some logging:

```
[16:34:32][D][micro_wake_word:355]: Detected 'Okay Nabu' with sliding average probability is 0.92 and max probability is 1.00
[16:34:32][D][media_player:074]: 'Media Player' - Setting
[16:34:32][D][media_player:078]:   Command: STOP
[16:34:32][D][media_player:087]:  Announcement: yes
[16:34:32][D][speaker_media_player:426]: State changed to ANNOUNCING
[16:34:32][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[16:34:32][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[16:34:32][D][ring_buffer:034]: Created ring buffer with size 9600
[16:34:32][D][speaker_mixer:310]: Starting speaker mixer
[16:34:32][D][speaker_mixer:318]: Started speaker mixer
[16:34:32][D][voice_assistant:528]: State changed from IDLE to START_MICROPHONE
[16:34:32][D][voice_assistant:535]: Desired state set to START_PIPELINE
[16:34:32][D][voice_assistant:222]: Starting Microphone
[16:34:32][D][voice_assistant:528]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[16:34:32][D][voice_assistant:528]: State changed from STARTING_MICROPHONE to START_PIPELINE
[16:34:32][D][voice_assistant:277]: Requesting start...
[16:34:32][D][voice_assistant:528]: State changed from START_PIPELINE to STARTING_PIPELINE
[16:34:32][D][voice_assistant:550]: Client started, streaming microphone
[16:34:32][D][voice_assistant:528]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[16:34:32][D][voice_assistant:535]: Desired state set to STREAMING_MICROPHONE
[16:34:32][D][voice_assistant:665]: Event Type: 1
[16:34:32][D][voice_assistant:668]: Assist Pipeline running
[16:34:32][D][voice_assistant:665]: Event Type: 3
[16:34:32][D][voice_assistant:679]: STT started
[16:34:32][D][light:036]: 'voice_assistant_leds' Setting:
[16:34:32][D][light:047]:   State: ON
[16:34:32][D][light:051]:   Brightness: 66%
[16:34:32][D][light:109]:   Effect: 'Waiting for Command'
[16:34:32][D][power_supply:033]: Enabling power supply.
[16:34:33][D][speaker_media_player:426]: State changed to IDLE
[16:34:33][D][speaker_mixer:323]: Stopping speaker mixer
[16:34:34][D][voice_assistant:665]: Event Type: 11
[16:34:34][D][voice_assistant:833]: Starting STT by VAD
[16:34:34][D][light:036]: 'voice_assistant_leds' Setting:
[16:34:34][D][light:051]:   Brightness: 66%
[16:34:34][D][light:109]:   Effect: 'Listening For Command'
[16:34:35][D][voice_assistant:665]: Event Type: 12
[16:34:35][D][voice_assistant:837]: STT by VAD end
[16:34:35][D][voice_assistant:528]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[16:34:35][D][voice_assistant:535]: Desired state set to AWAITING_RESPONSE
[16:34:35][D][voice_assistant:528]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[16:34:35][D][light:036]: 'voice_assistant_leds' Setting:
[16:34:35][D][light:051]:   Brightness: 66%
[16:34:35][D][light:109]:   Effect: 'Thinking'
[16:34:35][D][voice_assistant:528]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[16:34:35][D][voice_assistant:528]: State changed from AWAITING_RESPONSE to AWAITING_RESPONSE
[16:34:35][D][power_supply:033]: Enabling power supply.
[16:34:36][D][power_supply:033]: Enabling power supply.
[16:34:36][D][voice_assistant:665]: Event Type: 4
[16:34:36][D][voice_assistant:693]: Speech recognised as: " What time is it?"
[16:34:36][D][voice_assistant:665]: Event Type: 5
[16:34:36][D][voice_assistant:698]: Intent started
[16:34:36][D][voice_assistant:665]: Event Type: 6
[16:34:36][D][voice_assistant:665]: Event Type: 7
[16:34:36][D][voice_assistant:723]: Response: "12:34 PM"
[16:34:36][D][light:036]: 'voice_assistant_leds' Setting:
[16:34:36][D][light:051]:   Brightness: 66%
[16:34:36][D][light:109]:   Effect: 'Replying'
[16:34:36][D][voice_assistant:665]: Event Type: 8
[16:34:36][D][voice_assistant:745]: Response URL: "https://<redacted>/api/tts_proxy/4YZFijSUIJU3_iaxZPEzyQ.flac"
[16:34:36][D][voice_assistant:528]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[16:34:36][D][voice_assistant:535]: Desired state set to STREAMING_RESPONSE
[16:34:36][D][media_player:074]: 'Media Player' - Setting
[16:34:36][D][media_player:081]:   Media URL: https://<redacted>/api/tts_proxy/4YZFijSUIJU3_iaxZPEzyQ.flac
[16:34:36][D][media_player:087]:  Announcement: yes
[16:34:36][D][speaker_media_player:426]: State changed to ANNOUNCING
[16:34:36][D][voice_assistant:665]: Event Type: 2
[16:34:36][D][voice_assistant:764]: Assist Pipeline ended
[16:34:36][D][esp-idf:000][ann_read]: I (6446388) esp-x509-crt-bundle: Certificate validated

[16:34:36][D][ring_buffer:034][ann_read]: Created ring buffer with size 1000000
[16:34:36][D][speaker_media_player.pipeline:114]: Reading MP3 file type
[16:34:36][D][speaker_media_player:426]: State changed to IDLE
[16:34:36][D][voice_assistant:401]: Announcement finished playing
[16:34:36][D][voice_assistant:528]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[16:34:36][D][voice_assistant:535]: Desired state set to RESPONSE_FINISHED
[16:34:36][D][voice_assistant:528]: State changed from RESPONSE_FINISHED to IDLE
[16:34:36][D][voice_assistant:535]: Desired state set to IDLE
[16:34:36][D][light:036]: 'voice_assistant_leds' Setting:
[16:34:36][D][light:047]:   State: OFF
[16:34:36][D][light:109]:   Effect: 'None'
[16:34:46][D][power_supply:048]: Disabling power supply.
```

Note I tested the Response URL: "https://<redacted>/api/tts_proxy/4YZFijSUIJU3_iaxZPEzyQ.flac" from a device on the same network as the HA PE and it plays back fine.

Bit confused by this line:
[16:34:36][D][speaker_media_player.pipeline:114]: Reading MP3 file type

When the file type should be FLAC.

hagak, 2025-04-07:
Okay making progress.  Instead of playing the FLAC file in the browser I downloaded it.  And the file extension was mp3 which tells me that HA proxy is putting the wrong metadata on it.

hagak, 2025-04-07:
Yes when I look at the HTTP headers that is returned the Response header Content-Type: audio/mpeg which is wrong

floffaboff, 2025-04-11:
I had the same problem. Installing a proper /etc/mime.types file (e.g. pacman -S mailcap in arch linux) solved the issue in my case.

hagak, 2025-04-12:
Yep adding mailcap to my Dockerfile to have it install the /etc/mime.types solved my issue.  Thanks

closed 2025-04-12 by hagak