esphome/home-assistant-voice-pe #382: Voice PE gets stuck "responding"

Daverover66, 2025-04-10:
I have 5 Voice PEs and every day, at least one will get stuck "responding" after I ask it to do something.  The debug from the latest instance shows that everything appeared to work okay.  The intent was translated and executed and a response was apparently sent, but the device itself was stuck "responding" and no response was heard.

Here's a history trace from the device at the time.  I have an automation that triggers if the device gets stuck for 3 minutes and restarts the device.

[image]

HA Core 2025.4.0
OS 15.1
Supervisor 2025.3.04
Frontend 20250401.0

VPE Firmware: 25.3.4 (ESPHome 2025.3.3).  

This issue has been reported elsewhere in HA Core #142363

CraigStarCoder, 2025-04-18:
+1 Having exact same problem.

Core: 2025.3.4
Operating System:15.1
Hardware: Raspberry Pi 3 & Home Assistant Voice PE
Firmware: 25.3.4 (ESPHome 2025.3.3)

kahrendt, 2025-04-21:
Thanks for reporting! I haven't been able to reproduce it myself, but this does seem like a common issue spread across multiple users.

I'm hoping the just released latest beta firmware will fix it (25.4.1). I would appreciate any feedback with this beta on whether it fixes the issue or not!

Daverover66, 2025-04-22:
Thanks for the potential fix.  Do I need to "Take Control" of the device to install beta firmware?  I'll install this on one of our most used VPEs and see how it goes.

martijnrusschen, 2025-04-22:
Updated to the latest beta firmware, but still happening.

KennethLavrsen, 2025-05-02:
I experience this hanging blue ring almost once a day. I have 5 of then.
It happens in my case when I am doing announcements.

I have noticed that it always happens in connection with leaving home or arriving. In borh cases several announcements are being sent. My feeling is that it happens when you send multiple messages. It is like the new message prevents the previous to end. I can never make it happen with a single announcement. 

I have tried to change the setup so all announcements are sent by one script which is set to queue and only allow 1. That has not been enough.

I have had to make an automation that triggers when an assist entity has been in "responding" mode for more than 1 minutes.

Daverover66, 2025-05-02:
I'm not running the beta, but when  one of my VPEs got stuck responding I tried to send an Assist announcement to it and it failed.  Here's one of the log entries:

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/assist_satellite/entity.py:211
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 1 May 2025 at 20:57:28 (2 occurrences)
Last logged: 1 May 2025 at 20:58:41

[139962523283760] Error handling message: Unknown error (home_assistant_error) David Forrester from 127.0.0.1 (Mozilla/5.0 (Linux; Android 14; SM-T736B Build/UP1A.231005.007; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/135.0.7049.111 Safari/537.36 Home Assistant/2025.4.3-15918 (Android 14; SM-T736B))
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 28, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 816, in handle_execute_script
    script_result = await script_obj.async_run(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
        msg.get("variables"), context=context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1827, in async_run
    return await asyncio.shield(create_eager_task(run.async_run()))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 460, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    self._handle_exception(
    ~~~~~~~~~~~~~~~~~~~~~~^
        ex, continue_on_error, self._log_exceptions or log_exceptions
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 556, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 524, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1008, in _async_step_call_service
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 624, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/assist_satellite/entity.py", line 211, in async_internal_announce
    raise SatelliteBusyError
homeassistant.components.assist_satellite.errors.SatelliteBusyError
```
and here's another:

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/esphome/assist_satellite.py:410
[... 49 lines cut ...]
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/assist_satellite/entity.py", line 218, in async_internal_announce
    await self.async_announce(announcement)
  File "/usr/src/homeassistant/homeassistant/components/esphome/assist_satellite.py", line 350, in async_announce
    await self._do_announce(announcement, run_pipeline_after=False)
  File "/usr/src/homeassistant/homeassistant/components/esphome/assist_satellite.py", line 410, in _do_announce
    await self.cli.send_voice_assistant_announcement_await_response(
    ...<5 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/aioesphomeapi/client.py", line 1488, in send_voice_assistant_announcement_await_response
    resp = await self._get_connection().send_message_await_response(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<8 lines>...
    )
    ^
  File "aioesphomeapi/connection.py", line 838, in send_message_await_response
  File "aioesphomeapi/connection.py", line 820, in send_messages_await_response_complex
aioesphomeapi.core.APIConnectionError: Connection closed
```

bofisher, 2025-05-16:
Not sure if it helps, but I have this same issue on all 5 of my atom echos. The first conversation cycle after a fresh boot will always leave them stuck on Responding. Since you mentioned sending messages and announce, I thought I would give some details about my setup. I'm using external sonos media players to play any audio, as a result my firmware does not even config or define a speaker. No services or anything ever send it audio/messages.

I'm wondering if not configuring a speaker at all is the issue? Or hopefully that is a clue to the root cause.

```
i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO33
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

voice_assistant:
  id: va
  microphone: echo_microphone 
  noise_suppression_level: 2 #not used with microwake detection, but used with stream to HA
  auto_gain: 31dBFS #still referenced with microwake (default 31dBFS)
  volume_multiplier: 2.0 #still referenced with microwake (default 2.0)
  vad_threshold: 3 #not used with microwake
```

wambs, 2025-07-17:
I having this issue as well. I have 5 voice PEs. When doing an announcement to all 5, many times one of them will get stuck responding. The blue ring lights up and never stops. I have to pull the power to reset it.

omaramin-2000, 2025-08-08:
> I having this issue as well. I have 5 voice PEs. When doing an announcement to all 5, many times one of them will get stuck responding. The blue ring lights up and never stops. I have to pull the power to reset it.

I still have this issue with the Wyoming satellite to this day... IK nothing about ESPHome voice assistants as I don't use them.