esphome/home-assistant-voice-pe #385: Voice PE not playing any Media

dnwk, 2025-04-22:
Voice PE Firmware: 25.4.0 (ESPHome 2025.3.3) 
HAOS  Core 2025.4.3
Supervisor 2025.04.0
Operating System 15.2 

Recently my Voice PE stop playing any Media, Voice TTS or MP3.

This is the log on Voice when using Media → Text to speech → Piper for sending announcement

[16:14:57][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[16:14:57][D][esp32.preferences:114]: Saving 1 preferences to flash...
[16:14:57][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[16:15:02][D][media_player:074]: 'Media Player' - Setting
[16:15:02][D][media_player:081]:   Media URL: http://172.30.1.200:8123/api/esphome/ffmpeg_proxy/64ec6ad9ecdedbb26f81c74bc8f976be/4Qx4oyHA_nKszIU_dFZxWA.flac
[16:15:02][D][media_player:087]:  Announcement: yes
[16:15:02][D][speaker_media_player:426]: State changed to ANNOUNCING
[16:15:07][D][esp-idf:000][ann_read]: W (71640) HTTP_CLIENT: Connection timed out before data was ready!
[16:15:08]
[16:15:08][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[16:15:08][D][speaker_media_player:426]: State changed to IDLE

Here is the log on Voice when using media → my media → media file

[16:16:38][D][media_player:074]: 'Media Player' - Setting
[16:16:38][D][media_player:081]:   Media URL: http://172.30.1.200:8123/api/esphome/ffmpeg_proxy/64ec6ad9ecdedbb26f81c74bc8f976be/hAYVHvV4U_RzKkUnvXJqWA.flac
[16:16:38][D][speaker_media_player:426]: State changed to PLAYING
[16:16:43][D][esp-idf:000][med_read]: W (167502) HTTP_CLIENT: Connection timed out before data was ready!
[16:16:43]
[16:16:43][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[16:16:43][D][speaker_media_player:426]: State changed to IDLE

Both looks pretty identical on Voice PE logs.
I get these logs via https://web.esphome.io/ and then connect to my voice via USB
Those http media URL, I am able to download the file, if I manually visit it via browser.

Occasionally I may also see this error on HA side

Logger: aioesphomeapi.connection
Source: runner.py:154
First occurred: 4:14:15 PM (1 occurrences)
Last logged: 4:14:15 PM

home-assistant-voice-09df2c @ 192.168.1.39: Connection error occurred: [Errno 104] Connection reset by peer

But it’s not always happening.

nconder, 2025-05-05:
I'm having similar issues.  The response comes in to HA the home assistant voice device acts like it is playing something but no audio.

jdrowell, 2025-06-17:
OK so I found the solution to my problem in #378. It was a missing /etc/mime.types, which some Python package needs to supply the proper content-type headers for tts (probably the ffmpeg proxy). I'm running HA on FreeBSD (yes I know, not supported), which doesn't have that file. I copied it from a Linux host. voice-pe now talks to me again. Restart required of course.

tommyjlong, 2025-06-27:
Similar issue ...
My HAVPE works fine in general, but if if its been idle for a few days, and then I ask it say "what time is it?", I get:

```
[17:36:40.058][I][app:115]: ESPHome version 2025.5.1 compiled on May 28 2025, 16:11:55
[17:36:40.058][I][app:117]: Project Nabu Casa.Home Assistant Voice PE version 25.5.2
.....
[17:37:00.535][D][voice_assistant:456]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[17:37:00.535][D][voice_assistant:463]: Desired state set to STREAMING_RESPONSE
[17:37:00.535][D][media_player:074]: 'Media Player' - Setting
[17:37:00.535][D][media_player:081]:   Media URL: https://myhadomainport/api/tts_proxy/Mn-VUmK_CANp0kgHG1BebA.flac
[17:37:00.537][D][media_player:087]:  Announcement: yes
[17:37:00.538][D][speaker_media_player:408]: State changed to ANNOUNCING
[17:37:00.538][D][voice_assistant:598]: Event Type: 2
[17:37:00.550][D][voice_assistant:697]: Assist Pipeline ended
[17:37:01.520][D][esp-idf:000][ann_read]: I (877652666) esp-x509-crt-bundle: Certificate validated
[17:37:07.086][D][esp-idf:000][ann_read]: E (877658234) transport_base: esp_tls_conn_read error, errno=No more processes
[17:37:07.086][D][esp-idf:000][ann_read]: W (877658234) HTTP_CLIENT: Connection timed out before data was ready!
[17:37:07.086][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[17:37:07.086][E][speaker_media_player:328]: The announcement pipeline's file reader encountered an error.
[17:37:08.024][D][esp-idf:000][ann_read]: I (877659158) esp-x509-crt-bundle: Certificate validated
[17:37:13.544][D][esp-idf:000][ann_read]: E (877664700) transport_base: esp_tls_conn_read error, errno=No more processes
[17:37:13.550][D][esp-idf:000][ann_read]: W (877664700) HTTP_CLIENT: Connection timed out before data was ready!
[17:37:13.550][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[17:37:13.550][E][speaker_media_player:328]: The announcement pipeline's file reader encountered an error.
[17:37:14.926][D][esp-idf:000][ann_read]: I (877666080) esp-x509-crt-bundle: Certificate validated
[17:37:20.532][D][esp-idf:000][ann_read]: E (877671634) transport_base: esp_tls_conn_read error, errno=No more processes
[17:37:20.532][D][esp-idf:000][ann_read]: W (877671634) HTTP_CLIENT: Connection timed out before data was ready!
[17:37:20.532][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[17:37:20.532][E][speaker_media_player:328]: The announcement pipeline's file reader encountered an error.
[17:37:21.423][D][esp-idf:000][ann_read]: I (877672574) esp-x509-crt-bundle: Certificate validated
[17:37:26.994][D][esp-idf:000][ann_read]: E (877678136) transport_base: esp_tls_conn_read error, errno=No more processes
[17:37:26.994][D][esp-idf:000][ann_read]: W (877678136) HTTP_CLIENT: Connection timed out before data was ready!
[17:37:26.994][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_FAIL
[17:37:26.994][E][speaker_media_player:328]: The announcement pipeline's file reader encountered an error.
[17:37:27.008][D][speaker_media_player:408]: State changed to IDLE
[17:37:27.009][D][voice_assistant:329]: Announcement finished playing
[17:37:27.010][D][voice_assistant:456]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[17:37:27.012][D][voice_assistant:463]: Desired state set to RESPONSE_FINISHED
[17:37:27.019][D][voice_assistant:456]: State changed from RESPONSE_FINISHED to IDLE
```
So no audio is played.

If I retry this after a couple of minutes it usually works.

V3ritas1337, 2026-03-03:
Hi all, similar issue here with Claude API integration:

`2026-03-03 01:07:25.154 WARNING (MainThread) [homeassistant.components.esphome.manager] Home Assistant Voice 0a0f40: [W][http_media_source:403][HTTPRead_1]: Pipeline 1: HTTP request attempt 1 failed with status -1
2026-03-03 01:07:26.150 ERROR (MainThread) [homeassistant.components.esphome.manager] Home Assistant Voice 0a0f40: [E][component:468]: http_request cleared Error flag
2026-03-03 01:08:29.033 ERROR (MainThread) [homeassistant.components.esphome.manager] Home Assistant Voice 0a0f40: [E][http_request.idf:216][HTTPRead_1]: HTTP Request failed; URL: http://192.168.1.227:8123/api/tts_proxy/1jvQnHo7F15gS-kys_uDSw.flac; Code: -1
2026-03-03 01:08:29.037 ERROR (MainThread) [homeassistant.components.esphome.manager] Home Assistant Voice 0a0f40: [E][component:451][HTTPRead_1]: http_request set Error flag: unspecified
2026-03-03 01:08:29.038 WARNING (MainThread) [homeassistant.components.esphome.manager] Home Assistant Voice 0a0f40: [W][http_media_source:403][HTTPRead_1]: Pipeline 1: HTTP request attempt 1 failed with status -1
2026-03-03 01:08:30.031 ERROR (MainThread) [homeassistant.components.esphome.manager] Home Assistant Voice 0a0f40: [E][component:468]: http_request cleared Error flag`