esphome/home-assistant-voice-pe #391: Voice PE - hardware/molding issue causes mono playback with 3.5mm plug

whitty, 2025-05-01:
Apologies if this is the wrong place to log such an issue - this is physical issue with the Voice PE, and I could not find another place to log it, to ensure feedback gets to the right people before the next hardware revision.

An issue is discussed on the community Discourse:  https://community.home-assistant.io/t/voice-assistant-pe-voice-is-left-channel-only-music-is-dual-channel-stereo/837322

Description of Issue:

Mono playback can be heard when playing music through the 3.5mm connection.  You'll need music or sound files with clear left/right channels to demonstrate this.

The cause is summarised clearly [here](https://community.home-assistant.io/t/voice-assistant-pe-voice-is-left-channel-only-music-is-dual-channel-stereo/837322/2):

>  The plastic case of the VAPE has a hole that does not allow the connector housing of a 3.5 mm male to protrude inside. And female on the PCB inside is not close enough to the front so the receptable protrude. The result is that most 3.5 mm jacks do not properly connect.

The workaround proposed for users is to drill out the area as [shown here](https://community.home-assistant.io/t/voice-assistant-pe-voice-is-left-channel-only-music-is-dual-channel-stereo/837322/7)

Proposed solution:

The next revision of hardware should rework the plastic molding/recessing to allow the 3.5mm jack to be fully inserted.

kahrendt, 2025-05-01:
Thanks for reporting! More recently manufactured batches of the VPE have moved the 3.5mm connector closer to the edge of the PCB to minimize this issue.

closed 2025-05-01 by kahrendt

nikita-squirrel, 2025-05-29:
This is an inexpensive non-destructive solution for VPEs that have the 3.5mm jack socket connection problem. [https://www.aliexpress.com/item/1005003232102582.html](url)

I had the problem with a new VPE purchased mid May 2025 but I realised that it was a fat jack plug that was snagging on the case rather than matching the slim jack socket. As it would not seat properly, it was to be expected that it would only work through the left channel. I found this cable on AliExpress and now that it has arrived, can confirm that the plug is slim enough to match the socket and thus does not snag the case.  The VPE is now playing back through both channels on my B&W Zeppelin and sounds awesome. I am delighted. Not only are the plugs slender, but also the cable is very lightweight and flexible. It was also coiled nicely in its package so does not have the awkward stiff kinks of the fat cable I tried first, that had been folded back on itself and tied with a wire fastener.