esphome/home-assistant-voice-pe #393: Incorrect feature calculation in wakeword

erkamkavak, 2025-05-01:
Hello, 
I was testing a wakeword model on my computer with pymicro features and on the esp device with microwakeword component. I noticed that it seems like working much better on the computer. After a little inspection, I have seen there could be some errors on feature extraction calculation part of microwakeword. 

https://github.com/esphome/home-assistant-voice-pe/blob/86292b62c90cadaf6a6e777258699caaeef993b5/esphome/components/micro_wake_word/micro_wake_word.cpp#L176-L198

In the above part, based on the comments I think it shouldnt subtract INT8_MIN from the value but instead it should add it since INT8_MIN is -128. I thought this could be a simple error but I am not sure

kahrendt, 2025-05-01:
Thank you very much for reporting this! I've been investigating this carefully, and while you are correct about the sign, it thankfully isn't changing the actual features gnerated. With the current approach, we end up a value between 128 and 383 before clamping. However, since the clamp works on an int8 type instead of an int32 type, it is essentially ends up dropping the most significant bits before verifying it is within the bounds, which gets us to the equivalent number between -128 and 127.

To do this properly, it should be done like this:
```
value += INT8_MIN;
features_buffer[i] = static_cast<int8_t>(clamp<int32_t>(value_copy, INT8_MIN, INT8_MAX));
```

I've been comparing both approaches for several minutes with lots of different noise conditions, and it hasn't once ever had a difference between the two calculation approaches. I will still correct the code upstream in ESPHome so the code comment is accurate instead of relying on the clamp input type to fix it. (The micro_wake_word code in this repo will be removed after the next ESPHome release in favor of using the upstream code).

erkamkavak, 2025-05-01:
Thank you for the fast response. 
I am not sure about the calculations, probably something like you said is happening and it is not like all features are lost. But I feel like it is working better for me now. At least I can say that false negatives dropped visibly. I had to repeat 2-3 times before to trigger it but now it understands in the first try.

closed 2025-05-01 by kbx81