esphome/home-assistant-voice-pe #396: [REQUEST] Silence or Customize Wake‑Tone and Remap Single‑Tap Behavior

hannahmwright, 2025-05-05:
Every press of the Voice Assistant PE button triggers a loud chime, regardless of whether it’s a single‑tap, double‑tap, or long‑press. I use the button in my baby’s room to advance bedtime routines, and the wake‑tone regularly is startling and disrupts the white‑noise we keep playing.

A second limitation is that the single‑tap action is hard‑wired to launch the assistant, making it impossible to assign that tap to a custom automation.

I’d love two small tweaks that would make a huge difference. 

First, please let us **control the wake tone**: give us the **option to turn the chime on or off for each tap pattern** (single, double, or long press) along with a volume slider or a few softer built‑in sounds, or even the ability to upload our own gentle WAV or OGG file. 

Second, allow us to **repurpose the single tap**. It's currently locked to launching the assistant, but being able to map that tap to any automation, scene, or script would unlock many useful workflows.

grigi, 2025-05-12:
Adding to this, could we please have the ability to pass through single-clicks when the voice is NOT active. Having it as a way to cancel voice activation is very useful.

Reasoning:
I only ever use the built-in single click (on purpose) to cancel when the wakeword activated unwantedly.

By accident it happens often when I do a long-press slightly too short. Then I need to stop it, and wait a few seconds for it to give up on voice commands, THEN try again.

synesthesiam, 2026-10-01:
This can be changed in the firmware: https://github.com/esphome/home-assistant-voice-pe/blob/d4e6fa43d6a1a7d342b7d4d403567e06516719de/home-assistant-voice.yaml#L26