esphome/home-assistant-voice-pe #399: Home Assistant Voice repeats "No timers" when asking for timer state right when timer ends

TheLastProject, 2025-05-11:
# Description

When asking for the time left on the timer right before it ends, Home Assistant Voice starts repeating the sentence "No timers" instead of the "Timer finished" sound.

# How to reproduce
1. Say "Okay Nabu, set a timer for 10 seconds"
2. Wait until there are less than 5 seconds left
3. Say "Okay Nabu, how much time is left on the timer?"
4. Say "Stop" to make it end (once, it got stuck in the "responding" state forever, much like #382, but I couldn't reliably reproduce this, so that might be a coincidence)

I've recorded a video to make the issue more obvious than text would:
https://github.com/user-attachments/assets/35bb09bd-e12a-46ad-b759-8ac8c974b1ec

# Expected result
This is a bit of a tough one. I guess just say "No timers" once, while the "timer finished" sound plays repeatedly?

# Actual result
The repeating "timer finished" sound is replaced with the sentence "No timers"

TheLastProject, 2025-05-11:
I ran into this by pure coincidence checking up on a timer that just happened to expire. Most hilarious bug I've seen in a while :)

hecktech27, 2025-11-10:
i just ran into that bug as well. i can add that it will either play the response in a loop as in the video or if it was asked to perform an action in its area it will just play the confirmation sound in an endless loop. maybe that helps troubleshooting.