esphome/home-assistant-voice-pe #401: Hey Jarvis wake work does not work well

Tazmanian79, 2025-05-15:
I have 2 Home Assistant voice devices.

I want to use the wake word Hey Jarvis, but it's unreliable . If I try it 10 times, it only works 1 or 2 times.
When I use Hey Naby or Hey Mycroft, it works 9 times out of 10, even when I say it quietly.

I tried to use different pronunciations, even when I'm just next to the device, but it doesn't make any difference.

As the 2 other wake words don't have any problem, I think it's a bug in the Hey Jarvis wake word.

andreasbrett, 2025-05-15:
See https://github.com/esphome/home-assistant-voice-pe/pull/403

mirml, 2025-05-18:
Same problem, can't use "hey jarvis" wake word.

WhiteSockedDancer, 2025-05-21:
I also wanted to use "hey jarvis", but "okay nabu" works so much better! Is there a big gap in these two models performance?

binary100100, 2025-05-22:
Just ordered and set up three of them. 
I also chose "Hey Jarvis" because I already have an Alexa in every room and "Okay Nabu" sounds silly to visitors. I went through the process of generating my own custom wake words (Glados, Wheatley and HAL.)
I was hoping to use them with these. 😟
Anyway, "Hey Jarvis" only seems to work if I'm sitting 4-6 feet away from it. It's really hit/miss with it. I've enabled BETA firmware in hopes that this gets fixed promptly. The new firmware (25.4.1) seems to address an issue that I had encountered TWICE already where I would give it a command and it would just get stuck as 'playing' and not do anything else forcing me to unplug it.

closed 2025-11-05 by Tazmanian79