esphome/home-assistant-voice-pe #405: ERROR Installing ESPHome 2025.5.0

wendefeuer, 2025-05-22:
HI,
I get this Error than I try to install the new Version of ESPHome on the PE.

This is the Log:
[logs_home-assistant-voice-buero_run.txt](https://github.com/user-attachments/files/20380311/logs_home-assistant-voice-buero_run.txt)

UlrichThiess, 2025-05-22:
```
INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/buro-ug-voice.yaml...
INFO Updating https://github.com/esphome/home-assistant-voice-pe@dev
Failed config

external_components: [source /data/packages/dc43710d/home-assistant-voice.yaml:1530]
  - source: 
      type: git
      url: https://github.com/esphome/home-assistant-voice-pe
      ref: dev
    components: 
      
      Could not find __init__.py file for component micro_wake_word. Please check the component is defined by this source (search path: /data/external_components/3bf38338/esphome/components/micro_wake_word/__init__.py).
      - micro_wake_word
      - microphone
      - nabu_microphone
      - voice_assistant
      - voice_kit
    refresh: 0s
```

mdimjasevic, 2025-05-22:
I have the same issue and currently my Voice PE device is unusable since yesterday when I attempted to upgrade to ESPHome 2025.5.0.

It looks like something was not coordinated with the 2025.5.0 ESPHome release.

kahrendt, 2025-05-22:
Please take a look at the latest [yaml configuration](https://github.com/esphome/home-assistant-voice-pe/blob/dev/home-assistant-voice.yaml) and potentially update your own yaml if you have heavily modified it. We are now using the upstreamed versions in ESPHome instead of using external components in this repo.

There is sometimes a caching delay that seems to clear up after 24 hours, once we push a new firmware configuration here (which was done in the last day).

closed 2025-05-22 by kahrendt

mdimjasevic, 2025-05-22:
> Please take a look at the latest [yaml configuration](https://github.com/esphome/home-assistant-voice-pe/blob/dev/home-assistant-voice.yaml) and potentially update your own yaml if you have heavily [...]

Thank you, @kahrendt ! I will give it a try tomorrow and report back.

wendefeuer, 2025-05-22:
In the YAML I miss things like:
- ssid / password for WIFI
- api / encryption / key

where do I have to enter these?

mdimjasevic, 2025-05-24:
> Thank you, [@kahrendt](https://github.com/kahrendt) ! I will give it a try tomorrow and report back.

I just successfully built 2025.5.0 for my voice PE device. I am not sure if my cache was cleared before that, though. Is there a way for me to make sure I am building against the configuration with the fix you have pointed to, @kahrendt ?

mdimjasevic, 2025-05-24:
> In the YAML I miss things like:
> 
>     * ssid / password for WIFI
> 
>     * api / encryption / key
> 
> 
> where do I have to enter these?

These are settings specific to your setup. I do not enter the SSID and the password directly; rather, they are stored elsewhere and the configuration file just points to them, like so:

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
```

However, The API encryption key is listed directly in the file. It gets generated by ESPHome when you adopt the device in ESPHome Builder (assuming you use ESPHome that way within Home Assistant).