esphome/home-assistant-voice-pe #406: Unable to build EspHome adopted device since 2025.5 esphome update

FirstRulez, 2025-05-22:
I have noted the merge of #392  but since that merge I am not able to update my VoicePE device due to errors:

Build Log:
```
INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/home-assistant-voice.yaml...
INFO Updating https://github.com/esphome/home-assistant-voice-pe@dev
Failed config

external_components: [source /data/packages/dc43710d/home-assistant-voice.yaml:1530]
  - source: 
      type: git
      url: https://github.com/esphome/home-assistant-voice-pe
      ref: dev
    components: 
      
      Could not find __init__.py file for component micro_wake_word. Please check the component is defined by this source (search path: /data/external_components/3bf38338/esphome/components/micro_wake_word/__init__.py).
      - micro_wake_word
      - microphone
      - nabu_microphone
      - voice_assistant
      - voice_kit
    refresh: 0s
```
Configuration File:

```
substitutions:
  name: homeassistantvoice
  friendly_name: Home Assistant Voice

packages:
  Nabu Casa.Home Assistant Voice PE: github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  comment: ${friendly_name}

# Enable Home Assistant API
api:
  encryption:
    key: !secret esp_api_key2

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret local_domain
  fast_connect: on

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IP Address"
      id: ${name}_my_ip_address
    mac_address:
      name: "Mac Address"
      id: ${name}_my_mac_wifi_address
    ssid:
      name: "Connected SSID"
      id: ${name}_connected_ssid
    bssid:
      name: "Connected BSSID"
      id: ${name}_connected_bssid
    dns_address:
      name: "DNS Address"
      id: ${name}_dns_address
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true

sensor:
  - platform: wifi_signal
    name: "Wifi Signal"
    update_interval: 60s
    icon: mdi:wifi

  - platform: uptime
    name: "Uptime"
    update_interval: 180s
    icon: mdi:clock-outline

```

I can see the commit removes the external component reference for micro_wake_word, but that is still referenced in many places elsewhere in the home-assistant-voice.yaml file. Is that an error in the #392 commit or am I suffering from something else?

[image]

tbrasser, 2025-05-22:
Have you cleaned your build files?

FirstRulez, 2025-05-22:
When I try to clean build files I get the same error. Not sure why that makes any sense. Do I need to edit the config file removing the external package to allow the clean build files to work?

tbrasser, 2025-05-22:
You can also try: https://github.com/esphome/home-assistant-voice-pe/pull/403#issuecomment-2901680285

FirstRulez, 2025-05-22:
> Please see [#328 (comment)](https://github.com/esphome/home-assistant-voice-pe/issues/328#issuecomment-2669925244) and below

Thank you, this gave me enough pointers to work out how to effectively clean the bundles manually.

I have successfully updated now.

Thank you for the pointers

closed 2025-05-22 by FirstRulez