esphome/home-assistant-voice-pe #411: Feature request: Delay between wake sound and output

nero2503, 2025-05-28:
Hallo,
I'm using active speaker which go into stand by after some time. If the assist make an announcement it take some seconds until the speaker are ready, so the beginning of the announcement will be lost. It would be great to have the possibility to set a delay between the wake sound and the output of the message.

Thank you, BR Bernd

andreasbrett, 2025-05-29:
How do you deal with using Assist with the wake word in that setup? For the announcement you could create some workarounds to delay it but waiting for the speaker to come back from standby when you uttered the wake word would annoy the hell out of me.

nero2503, 2025-06-01:
So, that is not a problem, because the Assist starts listening even when the speaker is not ready. You just miss the wake sound. When you start a conversation by using the wake word it does not matter. It take at least some seconds until you phrase your sentence. When the answer comes, the speaker is already active. It's more a problem with announcements. E.g. an automation sends some information to the assist to phrase it to the user.