esphome/home-assistant-voice-pe #420: Increased false triggers after updating to firmware 25.5.2

weinshel, 2025-06-20:
I updated to firmware 25.5.2 a few days ago and have noticed an increased number of false triggers with the wake word detection even on the (default?) "slightly sensitive" sensitivity setting. The device will trigger even though there is no speech in the room or anywhere in the home. I have noticed this happening twice in the past day; the Whisper configuration I am using seems to transcribe the first message as "Oh!"; in one instance, the voice assistant even detected follow up conversation.

Let me know if there are logs I should share!

[image]

[image]

[image]

inertia666, 2025-06-26:
I have the same problem with the same firmware. Lowest sensitivty and Mycroft wake word. If the TV is on it is even worse and it keeps an open communication channel open and then tries to understand what the TV is saying.

If I shout "stop" it just ignores me so i end up sitting listening to it for several minutes until it gives up.

inertia666, 2025-08-25:
I am still getting daily false triggers caused by the TV on 25.7.0 and is increasingly frustrating.
I have a google nest too and it false triggers once in a blue moon (it is in the same room).