esphome/home-assistant-voice-pe #430: Request for precompiled .bin firmware for XU316-1024-QF60A

iMike78, 2025-07-19:
Hi everyone,

First of all, thank you for your outstanding work on the Voice PE project — it's an impressive piece of open hardware/software collaboration.

I'm currently working on a custom drop-in replacement PCB for the Google Nest Mini (Gen 2), based on your Home Assistant Voice PE architecture. The project integrates the XU316-1024-QF60A chip and is designed to be fully compatible with your firmware structure. You can find my open hardware project here:
👉 [MiciMike Google Nest mini v2 drop-in PCB](https://github.com/crime78/nest-mini-drop-in)

Over the past few days, I’ve been trying to build the firmware (.xe) from source using the provided repo (voice-kit-xmos-firmware), but I’ve run into persistent toolchain and CMake module issues — and unfortunately, the prebuilt files in the existing release don't include a .bin or flashable output file.

Would it be possible for you to provide (or generate) a precompiled .bin file targeting the XU316-1024-QF60A chip?
It would help tremendously with testing and firmware deployment for the community.

Thanks again for the great work — and let me know if I can support the project in any way from my side.

Best regards,
Imi
[@crime78](https://github.com/crime78)

kahrendt, 2025-07-28:
@kbx81 is more familiar with the XMOS binary stuff, but our yaml has a link to a binary we use if we need to auto-update the XMOS firmware:
```
voice_kit:
  id: voice_kit_component
  i2c_id: internal_i2c
  reset_pin: GPIO4
  firmware:
    url: https://github.com/esphome/voice-kit-xmos-firmware/releases/download/v1.3.1/ffva_v1.3.1_upgrade.bin
    version: "1.3.1"
    md5: 964635c5bf125529dab14a2472a15401
```
I'm not sure if that is exactly what you want, but I believe (its been awhile) that I flashed that particular file using DFU over USB at some point.

gnumpi, 2025-07-28:
> [@kbx81](https://github.com/kbx81) is more familiar with the XMOS binary stuff, but our yaml has a link to a binary we use if we need to auto-update the XMOS firmware:
>
> ```
> voice_kit:
>   id: [...]

@iMike78 , be aware that this bin file is an upgrade firmware image. This one only works if you have already a factory image running on your XMOS. Is this what you are looking for? Or are you interested in the factory image that also includes the boot loader? The factory image can be written directly to the spi flash memory starting from address 0x00.

closed 2025-07-29 by iMike78