esphome/home-assistant-voice-pe #438: Voice PE stuck "Enabling power supply"

EmilSepa, 2025-08-10:
After switching to Unifi APs my Voice PE is stuck in "Enabling power supply". 
I attached one of my log files below.
I am aware of the "fast connect" problems for ESPs and hidden SSIDs.
The Wifi AP is a Unifi U7 Pro with a IoT VLAN for Homeassistant and the Voice PE on 2.4GhZ. 

The different things I tried: 

- different software version for Voice PE: 2025.5.1, 2025.5.2, 2025.6.0, 2025.7.0
- hidden SSID and not hidden
- IoT-only-Mode on Unifi
- Took over control with ESPHome and activated "Fast connect"

I have read some blog posts about using specific wifi channels for esp32. 
Sadly, I don't have another AP to test this.
Before switching to Unifi, I've had a FritzBox and the Voice PE worked.

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/21706055/esp-web-tools-logs.txt)

Deathbyteacup, 2025-08-14:
I had this issue (also with UniFi APs, coincidentally) when using a USB power supply that couldn't provide enough amps. USB on a computer for example won't be enough. I switched to an Anker USB adapter and it resolved it. So I think it is a power draw issue.

EmilSepa, 2025-09-04:
After debugging my voice pipeline for a while, I found the error. 
It was never a power problem. My faster-whisper container was not working after changing my network setup for unraid. 
Now I'm confused: 
My piper is still working and reachable on the default network bridge (on unraid), my faster-whisper is not. 
I had to move it to a custom bridge with an extra IP.

closed 2025-09-04 by EmilSepa